makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #94: MKS Robin Nano WiFi Plugin  not WORKING !!!!!!!!!!!!!!!!!!!!!!!!

HulusiAKAR, 2020-09-10:
It has been 2 months since I bought the device and we are still waiting for an update. 
When will the device receive a Wifi update? We cannot use the wifi plug.
I have an error every time. We wouldn't buy this device if I knew that would happen.
Hear our voices now.

themacboy, 2020-09-10:
my god, why I read 4 diferent issues in 4 diferent sites about this same problem?

Please my time has a lot of value for me, try to don't make me lost my time reading same message 4 times.

All people know that Robin Nano v2 not support WI-FI yet ... time to time ...

anlkaplan, 2020-09-10:
@HulusiAKAR Merhaba sizde robin nano kullanıyorsunuz sanırım ben yeni aldım kurulum yapıcamda bir kaç soru sorma şansım var mı kurulumdan anlıyorsanız

HulusiAKAR, 2020-09-12:
> my god, why I read 4 diferent issues in 4 diferent sites about this same problem?
> 
> Please my time has a lot of value for me, try to don't make me lost my time reading same message 4 times.
> 
> All people know that Robin Nano v2 not support WI-FI yet ... time to time ...
> 
> [image]

You said your time is precious.
Is our time worthless?
It has been 2 months since I bought the device and we are still waiting for an update.
If your device cannot deliver the features it promises, you wouldn't have produced it.
Keep your promises.

HulusiAKAR, 2020-09-12:
> @HulusiAKAR Merhaba sizde robin nano kullanıyorsunuz sanırım ben yeni aldım kurulum yapıcamda bir kaç soru sorma şansım var mı kurulumdan anlıyorsanız

Olur anıl bunlardan ne köy olur ne kasaba.
Cihazı alalı 2 ay oldu hala güncelleme bekliyoruz. Cihazı aldığımda doğru dürüst çalışan bir firmware bile yoktu. Yok BLTOUCH çalışmaz, yok ekrana görüntü gelmez. Böyle şerefsizlik görmedim. Üstüne 70 tl para verip Wifi kartı aldık bir güncelleme getiremediler. Bir de çıkmış diyorki NanoV2 'de Wifi desteği yok. Ulan ******** ******* o zaman ne diye öyle reklam yapıyorsun.
[image]

HulusiAKAR, 2020-09-12:
> my god, why I read 4 diferent issues in 4 diferent sites about this same problem?
> 
> Please my time has a lot of value for me, try to don't make me lost my time reading same message 4 times.
> 
> All people know that Robin Nano v2 not support WI-FI yet ... time to time ...
> 
> [image]

[image] Yukarıda görülen Wifi Module Ne peki ********* ************

anlkaplan, 2020-09-12:
@HulusiAKAR Nasıl wifi özelliği yok anlamadım tanıtımları öyle kullanamıyor muyuz şuan YouTubeda videolarda kullanıyorlardı?Yeni montaj yapacaktım ondan bir kaç fikir alıyim dedim bende türk bulunca

HulusiAKAR, 2020-09-12:
Yok ya 2 aydır kendimi paralıyorum zar zor BLTOUCH güncellemesi getirdiler BLTOUCH çalışıyor şimdi çok şükür.
Kendilerince bir tema yapmışlar bir boka yaramaz Marlin 'nin orjinal temasına geri döndüm en azında o daha kullanışlı.
Kendi temalarında pek çok şeyi kapatmışlar neredeyse ekran üzerinden hiç bir düzenleme yapamıyoruz

HulusiAKAR, 2020-09-12:
@anlkaplan Takıldığın bir yer olursa özelden mesaj at en azından benim gibi vakit kaybetmemiş olursun. Bildiğim kadarıyla yardımcı olurum.

anlkaplan, 2020-09-12:
@HulusiAKAR Özelden mesaj atmayı daha çözemedim bu nasıl bir uygulamaysa 😂

TimCarrDP, 2020-10-14:
Marlin doesnt have this supported with the makerbase plug just yet.

Maybe if you add some more exclamation marks in the title post, it would help speed things up.

ahalekelly, 2020-10-14:
@Malderin seems to have gotten it working on 2.0 at least. See #99 and https://github.com/Jeredian/mks-wifi-plugin/issues/35
Any advice you can give?