makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #95: MKS Robin Nano v1.2 - Errors linking ELF

Nume1977, 2020-09-11:
### Description
Unable to compile, always generates errors when linking the elf for nano v1.2

### Steps to Reproduce

Just installed PlatformIO (clean install of latest version)
I downloaded the repository (zip), extracted it and changed 
default_envs = mks_robin_nano35 in platformio.ini

and changed these in configuration.h
#define MOTHERBOARD BOARD_MKS_ROBIN
#define TFT_480x320
#define TFT_LVGL_UI_FSMC 
#define TOUCH_SCREEN
  #define XPT2046_X_CALIBRATION 12316
  #define XPT2046_Y_CALIBRATION -8981
  #define XPT2046_X_OFFSET        -43
  #define XPT2046_Y_OFFSET        257

#### Additional Information
Compilation fails with multiple warnings:

```
Linking .pio\build\mks_robin_nano35\firmware.elf
.pio/build/mks_robin_nano35/src/src/lcd/touch/touch_buttons.cpp.o:(.bss.touch+0x0): multiple definition of `touch'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/tft_lvgl_configuration.cpp.o:E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/tft_lvgl_configuration.cpp:50: first defined here
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/SPIFlashStorage.cpp.o: In function `SPIFlashStorage::savePage(unsigned char*)':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/SPIFlashStorage.cpp:171: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/SPIFlashStorage.cpp.o: In function `SPIFlashStorage::loadPage(unsigned char*)':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/SPIFlashStorage.cpp:182: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferRead(unsigned char*, unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o: In function `update_spi_flash':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:247: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:249: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferRead(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:250: warning: undefined reference to `W25QXXFlash::SPI_FLASH_SectorErase(unsigned long)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:251: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:252: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o: In function `gCfgItems_init':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:153: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferRead(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:155: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferRead(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:185: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BlockErase(unsigned long)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:159: warning: undefined reference to `W25QXXFlash::SPI_FLASH_SectorErase(unsigned long)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:160: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:162: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:163: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:164: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:165: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o:E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:166: warning: more undefined references to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)' follow
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o: In function `update_gcode_command':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:258: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:260: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferRead(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:261: warning: undefined reference to `W25QXXFlash::SPI_FLASH_SectorErase(unsigned long)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:262: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:282: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o: In function `get_gcode_command':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:286: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:287: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferRead(unsigned char*, unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o: In function `gcode_preview(char*, int, int)':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:711: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:712: warning: undefined reference to `W25QXXFlash::SPI_FLASH_SectorErase(unsigned long)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:713: warning: undefined reference to `W25QXXFlash::SPI_FLASH_BufferWrite(unsigned char*, unsigned long, unsigned short)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ui.cpp.o: In function `Draw_default_preview(int, int, unsigned char)':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:911: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/draw_ui.cpp:954: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/pic_manager.cpp.o: In function `lv_get_pic_addr':
[... cut ...]
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/tft_lvgl_configuration.cpp:679: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/tft_lvgl_configuration.cpp.o: In function `tft_lvgl_init':
E:\Downloads\3D Printer\Mks-Robin-Nano-Marlin2.0-Firmware-master/Marlin\src\lcd\extui\lib\mks_ui/tft_lvgl_configuration.cpp:447: warning: undefined reference to `W25QXXFlash::init(unsigned char)'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_robin_nano35\firmware.elf] Error 1
```

doon3, 2020-09-13:
I thint #define TOUCH_SCREEN dont need...

Nume1977, 2020-09-13:
I have tried again without TOUCH_SCREEN, compilation successful! 
Thank you, this bug can be closed :)

closed 2020-09-13 by Nume1977