makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #97: Mks-Robin-Nano-Marlin2.0-Firmware doesn't load the configuration from robin_nano35_cfg.txt

molrc, 2020-09-17:
Hi all!

I just compiled the Mks-Robin-Nano-Marlin2.0-Firmware and I noticed it didn't load the configuration from robin_nano35_cfg.txt. Is this correct or am I missing something?

Thanks in advance!

### Description

### Steps to Reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior:** [What you expect to happen]

**Actual behavior:** [What actually happens]

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

ils15, 2020-09-17:
OLD marlin Firmware is based in IAR compile (closed), and use txt to configure file
now, MKS open firmware and you need to compile marlin with all configurations.
Cfg.txt will not work anymore
https://baizhongyun.cn/home/mkstoolview
you can use MKS TOOL to easy compile firmware, if you want

mks-viva, 2020-11-19:
This version firmware need not config txt file and it is change base on marlin firmware.

closed 2020-11-19 by mks-viva