makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #100: Firmware will not compile: "'pause_menu_response' was not declared in this scope"

Warringer, 2020-09-19:
I cannot find this problem anywhere...

I wanted to compile this firmware for my new Sapphire Plus, whuch uses the Robin Nano V1.2.

Now I have run into this problem.

`Executing task in folder Mks-Robin-Nano-Marlin2.0-Firmware: platformio run <

Processing mks_robin_nano35 (platform: ststm32@~6.1.0; board: genericSTM32F103VE; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103VE.html
PLATFORM: ST STM32 (6.1.1) > STM32F103VE (64k RAM. 512k Flash)
HARDWARE: STM32F103VET6 72MHz, 64KB RAM, 512KB Flash
DEBUG: Current (jlink) External (blackmagic, jlink, stlink)
PACKAGES: 
 - framework-arduinoststm32-maple 2.10000.200103 (1.0.0) 
 - tool-stm32duino 1.0.1 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SoftwareSerialM> 1.0.0
|-- <lvgl> 6.1.1
|-- <STM32ADC> 1.0
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_dialog.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_manuaLevel.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_manual_level_pos_settings.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_max_feedrate_settings.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_motor_settings.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_move_motor.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_number_key.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_operation.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_pause_message.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_pause_position.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_preHeat.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_print_file.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_printing.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_ready_print.cpp.o
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_set.cpp.o
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp: In function 'void btn_ok_event_cb(lv_obj_t*, lv_event_t)':
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp:152:9: error: 'pause_menu_response' was not declared in this scope
         pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
         ^~~~~~~~~~~~~~~~~~~
Compiling .pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_step_settings.cpp.o
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp:152:9: note: suggested alternative: 'pause_menu_disp'
         pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
         ^~~~~~~~~~~~~~~~~~~
         pause_menu_disp
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp:152:31: error: 'PAUSE_RESPONSE_EXTRUDE_MORE' was not declared in this scope
         pause_menu_response = PAUSE_RESPONSE_EXTRUDE_MORE;
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp: In function 'void btn_cancel_event_cb(lv_obj_t*, lv_event_t)':
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp:201:9: error: 'pause_menu_response' was not declared in this scope
         pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT;
         ^~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp:201:9: note: suggested alternative: 'pause_menu_disp'
         pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT;
         ^~~~~~~~~~~~~~~~~~~
         pause_menu_disp
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp:201:31: error: 'PAUSE_RESPONSE_RESUME_PRINT' was not declared in this scope
         pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT;
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/lib/mks_ui/draw_dialog.cpp:201:31: note: suggested alternative: 'ADVANCED_PAUSE_RESUME_PRIME'
         pause_menu_response = PAUSE_RESPONSE_RESUME_PRINT;
                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~
                               ADVANCED_PAUSE_RESUME_PRIME
*** [.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_dialog.cpp.o] Error 1
Marlin/src/lcd/extui/lib/mks_ui/draw_pause_message.cpp: In function 'void lv_draw_pause_message(PauseMessage)':
Marlin/src/lcd/extui/lib/mks_ui/draw_pause_message.cpp:38:10: error: 'PAUSE_MESSAGE_PAUSING' was not declared in this scope
     case PAUSE_MESSAGE_PAUSING:  clear_cur_ui(); lv_draw_dialog(DIALOG_PAUSE_MESSAGE_PAUSING); break;
          ^~~~~~~~~~~~~~~~~~~~~
Marlin/src/lcd/extui/lib/mks_ui/draw_pause_message.cpp:38:10: note: suggested alternative: 'PAUSE_MESSAGE_PARKING'
     case PAUSE_MESSAGE_PAUSING:  clear_cur_ui(); lv_draw_dialog(DIALOG_PAUSE_MESSAGE_PAUSING); break;
          ^~~~~~~~~~~~~~~~~~~~~
          PAUSE_MESSAGE_PARKING
*** [.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_pause_message.cpp.o] Error 1
========================================================================================================================================================== [FAILED] Took 5.15 seconds ==========================================================================================================================================================

Environment       Status    Duration
----------------  --------  ------------
mks_robin_nano35  FAILED    00:00:05.153
===================================================================================================================================================== 1 failed, 0 succeeded in 00:00:05.153 =====================================================================================================================================================
The terminal process "platformio 'run'" terminated with exit code: 1`

Is there any fix out there?

imberest, 2020-09-19:
Have you messed up with UI? If not, try compiling stock one. I am playing with it and it compiles.

Warringer, 2020-09-19:
I will need to try that...

Warringer, 2020-09-19:
The problem is that I got a Robin Nano 1.2, with the TFT_LVGL_UI_FSMC display, not the TFT_LVGL_UI_SPI display.

imberest, 2020-09-19:
So you uncommented that TFT?

Warringer, 2020-09-19:
Yes.

I tried with a clean pull and just uncommented that display and commented the other. No dice.

appliedmike, 2020-09-19:
I have a Robin nano 1.2 and "FSMC" display but I use "#define TFT_480x320" not TFT_LVGL_UI_FSMC as I get the same error
This gives me full screen display, Ive tried to compile with the FSMC  drive to gain LVGL but I get Ram overflow issue :-(

makerbase-mks, 2020-09-26:
@Warringer If you use the lvgl interface, you need to disable ADVANCED_PAUSE_FEATURE

rouhit110, 2022-06-22:
> @Warringer If you use the lvgl interface, you need to disable ADVANCED_PAUSE_FEATURE

Unfortunately, I have exactly the same problem
I'm using the ts35 and need fes
I can no longer compile when I enable FILAMENT_RUNOUT_SENSOR
Why should I turn them off when I need them?
Should this problem be solved by your support team or Marlin?