makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #98: Thermal Runaway

imberest, 2020-09-19:
I have a bit of a headache with my Anycubic i3 Mega which originally had a Trigorila 32-bit board.

I decided to swap it because of loud A4988 drivers and got myself an MKS Robin Nano 1.2.

This printer is heavily modded. All the hot-end is swapped to MT2.

Now, MT2 mod was done with Trigorilla, so no problems back then. Also, I know that Anycubic locks their max fan speed at ~80 %, so with that in mind done tests with 100/80/50/30 % cooling fan and the result is the same; temperature drops, and I get "Thermal Runaway".

Now, the strangest part is: If I set manually 200 °C, wait for it to climb, and then blast 255 °C on the cooling fan, it drops to 199 °C and then back to 200 °C.

PID autotune was performed numerous times, with values saved, PID repeated, new values stored, etc.

Anycubic i3 Mega
MKS Robin Nano 1.2
E3D V6 original.
Marlin 2.0.5.3
And snippets from Marlin cfg.

```
#define TEMP_SENSOR_0 5

// Comment the following line to disable PID and enable bang-bang.
#define PIDTEMP
#define BANG_MAX 255     // Limits current to nozzle while in bang-bang mode; 255=full current
#define PID_MAX BANG_MAX // Limits current to nozzle while PID is active (see PID_FUNCTIONAL_RANGE below); 255=full current
#define PID_K1 0.95      // Smoothing factor within any PID loop

// Ai3 mega
 #define DEFAULT_Kp 27.69
 #define DEFAULT_Ki 2.02
 #define DEFAULT_Kd 94.89
```

I have XYZ cube and fan on from layer 3 gcode. Always fails on layer 3, so turned manually fan from layer one, was printing fine till layer 3 and then thermal runaway. So done a test, sent gcode to see if M106 is an issue.

```
M104 S210 T0
M109 S210 T0
M106 S204
```
But no, printer kept temperature steady... Makes me crazy, please help

maalouc, 2020-09-19:
Perform the autopid with the Part cooling Fan turn on at full speed. Have an Ai3M also, that works for me!

makerbase-mks, 2020-09-19:
@imberest 
You should ensure that the pid parameter in your reconfiguration.h is effective. You can try to execute the eeprom setting of the motherboard, first use M502, and then execute M500. In addition, should you increase the temperature protection time? Or change the adjustment range of pid, for example #define PID_FUNCTIONAL_RANGE 40

imberest, 2020-09-19:
@maalouc 
Yes, I know that and did it numerous times. even did when the nozzle is almost touching the bed, as there should be most cooling on the nozzle. Haven't solved the issue.

@makerbase-mks 
I will give it a try, but that doesn't explain why manually setting temp and fan doesn't break it up? 
Also, the last part, where I set the cooling fan manually, on the start of the print, and it started dropping when it reached layer 3, where gcode has `M106 S204`.

imberest, 2020-09-20:
@makerbase-mks Still no good, temp drops 10C and then stays kind of [Temp image](https://i.stack.imgur.com/gIYAG.png)

themacboy, 2020-10-18:
use mosfet to avoid overload on mother board. Thas solves a similar problem on my nano.

3D-N00B, 2023-03-18:
Hi i have same problem with BED temp.
tried with Mosfet, a new Mosfet (booth from MKS) and direct from board, 
if i set temp to 60, when reach 51/52 BOARD stops send signal to bed, was 12V steady, then drops to 2V~3V and Thermal Runaway occur