makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #99: SD card problem

Malderin, 2020-09-19:
Hello!
Mks robin nano v.2 board The latest firmware version from this branch is installed.
The problem with displaying the contents of the memory card if files are saved on it via the mks wifi plugin as TFT. When entering the memory card through the menu, the interface freezes. Only reboot helps. MKS wifi plugin installed from cura store.
On the latest marlin bugfix 2.x, everything works fine.

nehilo, 2020-09-21:
Ihave this problem too!When you fix it?

Malderin, 2020-09-21:
Если плагин установлен из маркета куры, то есть такая беда. Если плагин установить отсюда:
https://github.com/Jeredian/mks-wifi-plugin
то по крайней мере при заходе на карту памяти принтер не виснет. Однако все миниатюры разных моделей будут одинаковые (от какой то одной модели). При печати отображение миниатюр будет нормальное.
Разработчик плагина по ссылке выше сказал что проблема на стороне прошивки, так что как ни крути, вопрос сюда, в эту ветку.
См. так же
https://github.com/Jeredian/mks-wifi-plugin/issues/35
--
Отправлено из Mail.ru для Android понедельник,

velkhor, 2020-09-22:
Hi. I have board v1.2, but have similar problem. I find a simple solution:
step 1. remove SD card
step 2. wait 1-2 sec.
step 3. insert SD card
step 4. printer should continue, if freeze back to step 1 ;)

Give me know if this help.

Malderin, 2020-09-23:
> Hi. I have board v1.2, but have similar problem. I find a simple solution:
> step 1. remove SD card
> step 2. wait 1-2 sec.
> step 3. insert SD card
> step 4. printer should continue, if freeze back to step 1 ;)
> 
> Give me know if this help.

Checked it out.
This doesn't work.

Artemasd1974, 2020-09-24:
Здравствуйте!  Та же проблема, вам не удалось что-то выяснить по этой проблеме?

Malderin, 2020-09-24:
Рекомендую прочитать сперва это:
https://github.com/Jeredian/mks-wifi-plugin/issues/35#issuecomment-695202204
Вообще проблема не решена. Для себя пока поставил плагин из ветки по ссылке выше, с ним конечно показываются одинаковые миниатюры для всех моделей на карте памяти, но по крайней мере принтер не виснет при заходе на карту.

Artemasd1974, 2020-09-24:
По ссылке читал прежде. Спасибо.

velkhor, 2020-09-24:
I resolve my problem (board 1.2). I think there is some bug when flashing firmware to eprom.
When i put all files to upload (robin, wifi, lcd....), afret flash printer freeze. After that i copy to SD card only Robin_nano35.bin and upgrade again.
Everything looks ok, no freezes.
You can download software again (Robin_nano35.bin)
Or you can tru compile own vesrion of firmvare only few changes need.

closed 2020-11-08 by Malderin