makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #102: Add/Fix Support for M73 - LCD_SET_PROGRESS_MANUALLY

paintballkev812, 2020-09-25:
Enabling #define LCD_SET_PROGRESS_MANUALLY results in an error during the sanity check. mks_ui/draw_dialog.cpp makes reference to this variable, but seems it is either not fully implemented, or there is an error.

makerbase-mks, 2020-09-26:
Hello, I am a little confused, when does M73 need to be used? If it is set to 50%, is it similar to POWER_LOSS_RECOVERY function?

paintballkev812, 2020-09-26:
M73 will manually override the print progress displaying on the printer LCD. Some slicing programs will embed M73 commands into the gcode to provide more accurate percent completed and time remaining estimations. Additionally, a few octoprint plugins use this command (along with M117) to update the printers display information when the print is being sent from octoprint.

M73 P<percent completed> R<Time Remaining in Minutes>
M73 P8 R135 //Would notate 8% complete, 135 minutes remaining.

It looks like this PR added the logic for the display, but it cannot compile with needed flags enabled so I think logic needs to be added for the sanity check area. https://github.com/MarlinFirmware/Marlin/commit/c5bfdb0a7229e2fff4c8eb20fff0efadc71c42cb