makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #106: Robin nano v1.2

doon3, 2020-09-27:
firmware often freezes before printing....

MS1987, 2020-10-06:
@doon3 , what is the detail? How about the temperature of the heater? Has it reached the target value and been stable?

TimCarrDP, 2020-10-14:
The LCD doesnt show any halt errors, so its possible you have a hardware issue and dont know it. Run it via USB and something like Repetier host will automatically let you know when theres a fault.

doon3, 2020-10-14:
the error occurs when I heat up the table before printing ...

TimCarrDP, 2020-10-14:
Quite a few things it could be mate, We really need more information.

But most common would be its a HALT because of an thermistor setting/hardware issue, or incorrect WATCH_TEMP_PERIOD/WATCH_TEMP_INCREASE settings.

Hook the board up via USB and connect to it in Repetier host, then post the errors, and serial communications that appear there.

doon3, 2020-10-14:
I understand you. my printer is currently disassembled. as soon as I put it together I will try and let you know. I hope it will work out this week

doon3, 2020-10-17:
two days trying to catch the error with Repetier host - can't catch... if i succeed i will post it here

mks-viva, 2020-12-30:
@doon3 Haha! You can update firmware to the latest version and test it.

closed 2021-09-02 by mks-viva