makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #107: Any available GCODE to have LCD display information when printing via usb? (eg. with Cura)

mgcrea, 2020-10-01:
### Description

I currently print a lot via USB, and the screen does not update when I start the print (pretty normal).

Is there any GCODE I could add to the pre-print GCODE so that I can have a print status update on the LCD screen?

Ideally would be awesome to have the same screen than when printing via SD but not sure if possible.

### Info

I'm running stock Marlin v2.0.7 on an FLsun Q5 modded with an MKS Nano V2.

Thanks!!

ahalekelly, 2020-10-06:
Seconded! It should show the same screen as printing from SD. There's an M117 gcode in Marlin to set an LCD message but it doesn't do anything on this version.

TimCarrDP, 2020-10-14:
This isn't really an issue for github. Maybe the marlin facebook forums might help. Not even maker base specific, so you might have luck there.
I vote this gets closed.

paintballkev812, 2020-10-16:
This is a duplicate of #102 . Cura uses M73 to update printer status (same as octoprint). Support for M73 (as well as M117) would be needed in addition to access to the 'printing screen'.

paintballkev812, 2020-10-16:
I should also mention, it is possible to use stock marlin with the 'Touch UI' to regain this functionality. You will lose the current UI and will instead be replaced with the marlin UI that has been adapted for these touch LCDs.
[image]

mgcrea, 2020-10-19:
@paintballkev812 haven't tried stock marlin yet, do you just need to comment out `//#define TFT_LVGL_UI_SPI` or is there a specific build flag to enable to have a working stock touchscreen?