makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #108: [Question] [Robin Nano 2.0] Differences with official Marlin?

mgcrea, 2020-10-01:
Now that most of the LVGL work has been merged back into the official Marlin repo, are we supposed to switch to using the official Marlin now? What are the differences?

I'm running a modded FLsun Q5 (with MKS Robin Nano V2) with the official Marlin v2.0.7 and everything works great.

Really awesome that you guys managed to merge back your work into official Marlin. Really cool to know that your boards will be somehow supported for a very long time. Very good move IMO. Thanks!!

makerbase-mks, 2020-10-06:
@mgcrea At present marlin V2.0.7 version has merged all the functions of robin nano, you can use it directly. Of course, if there are any subsequent changes, we will modify it in this version first, and then submit it to malrin, waiting for the merger... Thank you for your attention.

simpat1zq, 2020-10-21:
Hi. If I go to the official marlin and want to compile for the nano 1.2 board, which config file would be best to use?

closed 2021-02-21 by MKS-Sean

damianCooper, 2021-09-27:
> Now that most of the LVGL work has been merged back into the official Marlin repo, are we supposed to switch to using the official Marlin now? What are the differences?
>
> I'm running a modded FLsun [...]

I would be interested in the exact config files that you have as I keep running into issues trying to get my Q5 running with the robin nano v2