makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #115: still does not work transferring file using cura

JinwooJang, 2020-10-16:
### Description

environment
 - board : Robin nano v2.0
 - firmware : built by latest source code(with changes "Fix the problem of cura wifi-plugin transferring files through wifi module")
 - PC tool : Cura 4.7.1 + bug fixed mks-wifi-plugin for cura 4.7.1

monitoring and running GCODE via wifi is ok.
but transferring file still failed at progress 20~30%.

have you success when you test ?
is there any setting to enable transferring file ?

#### Additional Information
[marlin_config.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5392474/marlin_config.zip)

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Malderin, 2020-10-17:
I confirm all of the above. I hope the file transfer will be completed.
Want to add. If there is already gcode on the memory card, it cannot be run from cura either.

MKS-Sean, 2020-10-19:
I tested it in cura4.5, but I think this problem has nothing to do with the version. I will use a larger file to test it.

themacboy, 2020-10-23:
Not working for me too. Brokes after 7%. 

Will updating I see strange characters in the top left of the screen.

JinwooJang, 2020-10-29:
I'm checking latest build. build error was fixed.(qr_encoder.h)

JinwooJang, 2020-10-30:
it works!!
thanks.