makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #119: New LVL causing SD card issues

TimCarrDP, 2020-10-24:
Something in the updated LVL library is causing the printer to lock up once selecting a print from SD.
It simply loads, screen blinks, but then all the target temps show "0" an the printer sits idle. Nothing to report over serial.
It hasnt crashed, as the display is still responsive and reading out ambient temps.

TimCarrDP, 2020-10-24:
Corrupt download of the library. Deleting and recompiling fixed the issue.

TimCarrDP, 2020-10-24:
Closed too soon. Issue still exists.

daniel78ktn, 2020-10-31:
I have the same problem, about 2 or 3 times it works, then the same problem like TimCarrDP.
Restore the Eprom default in LVGL menu, doses not work. I´ve don´t test the workaround with marlin stock UI. 
I have Robin Nano V2.0 Mainboard. Thanks

MKS-Sean, 2020-10-31:
@TimCarrDP @daniel78ktn 
Is it a card compatibility issue? I think it failed to read the card and did not read the gcode instruction. You can try to format the sd, and test whether it improves

daniel78ktn, 2020-10-31:
I have used another SD-Card (witch was tested bevore) same problem.
When i click a Gcode to print  Menue seems to be double loaded then are the buttons Resume, Stop , and Option aviable. 
When i click Resume , Stop Button don´t come, alltimes Resume Button.
No Print will started.

daniel78ktn, 2020-11-03:
There must be a problem with Gcode Preview in the Printing Menu. I have flashed a new  compiled firmware, then i print a gcode programm from sd card, same problem print menu double loaded, only resume, stopp, and option buttons are show.
Then i reboot the printer, go to the printing menu, select a gcode with preview picture, and started it.
The menu show a preview picture from another gcode that is **not** on the actualy insert sd card and was **not** loaded after flashing the new firmware. Now you can see how the new Previewpicture over the old one overlayed. The Same problem no print will started.
I hope my observation will help you to find the bug.

X-Ryl669, 2020-11-25:
I'm experiencing somehow the same issue. I have a SD card with a gcode produced by Cura with the MKS Wifi plugin, and it embeds a picture. As soon as I try to list the files from the print menu, it crashes.

X-Ryl669, 2020-11-26:
By reading the last change in Marlin, I've seen they have added a lot of `watchdog_reset()` in the lengthy LVGL calls. I'm wondering if the crash/freeze here are related to the watchdog killing the system while it's decoding pictures. Thus, it might be interesting to either upgrade to Marlin 2.0.7.2 version (with the included fixes except for GCode image loading) or re-apply their [commit](https://github.com/MarlinFirmware/Marlin/commit/f7f1224941ef0220603106f4a1b6b4ceabdfd196) and extend for gcode picture extraction.

X-Ryl669, 2020-11-26:
If you don't have watchdog enabled, please ignore the previous comment, obviously ;-)

X-Ryl669, 2020-11-26:
Can you try to:
1. Format the SD card (so there is no file left)
2. Put a single gcode file that fails
3. Reproduce the issue
4. Open the file on the sdcard on your computer with a text editor and remove all the top lines containing the picture (from `;simage;` to the line **before** `;Marlin`)
5. Save and plug the SD card back in the printer
6. Check if the issue is gone and report

Marterus, 2020-12-29:
I have the same problem.
If there are no TFT files on the SD card, then everything works fine. But if at least one TFT file appears, the system crashes.

X-Ryl669, 2020-12-29:
I've posted a PR in the bugfix-2.0.x branch to solve this. It's probably also solved in official Marlin's bugfix-2.0.x branch although differently.

Marterus, 2021-01-20:
I checked it on several SD cards, the problem is only for 16GB and 32GB. Everything works fine with 128MB and 256MB cards.

X-Ryl669, 2021-01-20:
It's indeed fixed in the official Marlin's firmware with `bugfix-2.0.x` branch.

WorkFlowOne, 2022-05-08:
I have the same problem and if the gcode file in the SD's root, it can be printed.