makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #120: WIFI transfer

Malderin, 2020-10-30:
### Description
1.  Printing does not start automatically after file transfer;
2. It is impossible to start printing an existing file on a memory card through the Cura;
3. File transfer is unstable. Not every file is transferred over Wi-Fi.

Malderin, 2020-10-30:
For information:
This bug https://github.com/MarlinFirmware/Marlin/issues/19914 is missing in this firmware.

MKS-Sean, 2020-10-31:
@Malderin 
I tested the stability in cura4.5, I can transfer 200M gcode files, and there are problems. If your transmission often makes mistakes, I suggest formatting the SD card. I will continue to test it in other versions.

Malderin, 2020-10-31:
> @Malderin
> I tested the stability in cura4.5, I can transfer 200M gcode files, and there are problems. If your transmission often makes mistakes, I suggest formatting the SD card. I will continue to test it in other versions.

Try it like this:
1. Transfer several small files (2-4 Mb);
2. Transfer 1-2 large files (15-20MB)
3. Steps 1-2 should be done without rebooting the board.

Usually an error appears on item 2.
The BLTouch error described above appears after transferring any file.

Malderin, 2020-10-31:
> * Printing does not start automatically after file transfer;
> * It is impossible to start printing an existing file on a memory card through the Cura;

@MKS-Sean fixed, Checked, fixed

Malderin, 2020-10-31:
> 5\. After transferring the file via Wi-Fi, the bltach stops working. If, for example, you send the axes to "all home", the z-axis probe does not extend.

@MKS-Sean , see video: https://yadi.sk/i/TTtMszwsFePL1g
After a successful file transfer via Wi-Fi, the probe probe does not extend. X and Y axes move with strong vibration

Malderin, 2020-10-31:
@MKS-Sean , @makerbase-mks ,
Found a new bug.
1. The position of the "yes / no" switch in the setting / config / advanced setting / wi-fi parametr setting / Cloud menu item does not change
2. If you select "no", there will be a black screen in the wi-fi menu

Video: https://yadi.sk/i/uCu7PNm2QXMMmQ

MKS-Sean, 2020-11-02:
> Try it like this:
>
> 1. Transfer several small files (2-4 Mb);
> 2. Transfer 1-2 large files (15-20MB)
> 3. Steps 1-2 should be done without rebooting the board.
>
> Usually an error appears on item 2.
> The [...]

Yes, I tested it like this:
1. Transmit the 3M model 5 times
2. The 15M model is transmitted 5 times
Can succeed both times!

Malderin, 2020-11-10:
@MKS-Sean , 
The problem has been resolved.
I did the following:
1. Slightly increased the buffer (perhaps it is not necessary);
2. Increased the number of connection attempts and the interval;
3. Picked up a better quality memory card.
All that I changed can be viewed here:
https://github.com/Malderin/Robin-nano-V2-kingroonKP3/commit/43f3f7901575423ed27f7252a8fe4c5d7be80161
Did a lot of tests, no more problems.

closed 2020-11-10 by Malderin

ysnuysl, 2023-12-27:
How were you able to send the image of the part you produced to the LCD screen? I can not do this.