makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #121: MKS Robin Nano V2 - nothing happens when printing is started

AdamMulica, 2020-11-03:
Hello, All options in the program when run manually work like ALL HOME, exstruder, hotend, hotbed etc. When I upload a file for printing, nothing happens and it goes to the resume
[image]

When I set the paramatra manually and click resume, nothing happens
[image]

What can I do wrong? I am using the CURA program and the settings for Alfawise U30. I don't know what settings I have to use for starting and ending printing?

Malderin, 2020-11-03:
if you have a filament runout sensor, disable it. It doesn't work in this firmware, I had such a problem with this.

AdamMulica, 2020-11-03:
#define FIL_RUNOUT_ENABLED_DEFAULT true I have to enter here "false"

daniel78ktn, 2020-11-03:
I have exactly the same problem, wrote in the post: SD card issues #119

Malderin, 2020-11-03:
> #define FIL_RUNOUT_ENABLED_DEFAULT true I have to enter here "false"

should be so
//#define FILAMENT_RUNOUT_SENSOR

daniel78ktn, 2020-11-03:
I have //#define FILAMENT_RUNOUT_SENSOR  in config.h otherwise i get compile error.

AdamMulica, 2020-11-03:
I have //#define FILAMENT_RUNOUT_SENSOR and don't work :(
[image]

Malderin, 2020-11-03:
> I have //#define FILAMENT_RUNOUT_SENSOR and don't work :(
> [image]

Do you have a filament runout sensor? Disconnect the cable from it

AdamMulica, 2020-11-03:
Yes a i have a filament runout sensor. I unplugged the cable and then nothing.

daniel78ktn, 2020-11-03:
I am sure we have tow bug´s. @Malderin you are right. Disconnect filament runout sensor will help but now printer won´t heating up bed and nozzle. Pause Button is now on display. I have connected Filament sensor again but don´t feeding filament inside. Now printer heat up and start. There must be High / Low signal in a config to change? 
Second bug: I´ve you are press Stop on the Display to cancel the print, it won´t clear in memory cause i can only start the print which was stopped bevore. Another gcode won´t start the printer!

AdamMulica, 2020-11-04:
hello, @Malderin I unplugged the cable from the motherboard. The PAUSE option appeared as he described it @daniel78ktn but... 
[image]

...does not heat the table and hotend. Time flies but nothing happens :(

AdamMulica, 2020-11-04:
When i use printed from USB this work...
[image]
[image]

daniel78ktn, 2020-11-04:
hmm.... printing via WIFI works also, at the moment.....   SD-Card printing works via  #define TFT_480x320_SPI

AdamMulica, 2020-11-05:
@daniel78ktn I changed 2 parameters:
#define TFT_480x320_SPI
and
#define USE_WIFI_FUNCTION   0
I had to set wifi to 0 because an error popped up during compilation and now the printer prints normally from the SD card.
Thank you very much for help.
As for the //#define TFT_LVGL_UI_SPI (version with icons), I will wait for the working firmware
[image]
[image]

daniel78ktn, 2020-11-05:
This Info was from TimCarrDP     New LVGL causing SD card issues #119

rcstar6696, 2020-11-24:
Issue still persists for me. Tested today

susmaryosep, 2021-01-31:
I started getting this issue, using the current firmware... then I see ver3 is out? I just got this v2... seems like they are much interested in selling product and making money rather than fix the issue. they just let the users troubleshoot on their own and post the solution/fix... 

Anyhow, I'm gonna try switching to SPI. see if that works.

AdamMulica, 2021-02-12:
Hello,
I downloaded all the content from makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware.

[image]

I compiled everything from scratch.
And as you can see it works. Prints without problems. I only have to check the filament detection. Every option in the firmware works as it should. It finally works!

Sergey-70, 2021-02-16:
AdamMulica, добрый день. Не могли бы Вы  скинуть скомпилированную прошивку на Github? Я не могу никак победить свою. Заранее спасибо.

AdamMulica, 2021-02-16:
@Sergey-70 what's your printer model? enter your e-mail address, I will try to help.

Sergey-70, 2021-02-17:
У меня принтер CoreXY Elf 3D. Email: sergroza@mail.ru. Спасибо.

MKS-Sean, 2021-02-21:
Sorry for being late, any questions? Hope i can help

m63445, 2021-06-27:
Can some one send me a bin file for mid robin nano 2.0 with a ts35 screen to my email Thesedores@aol.com. I can’t get my printer to heat up at all. Thanks so much

AleefRibeiro, 2021-11-27:
> @daniel78ktn I changed 2 parameters: #define TFT_480x320_SPI and #define USE_WIFI_FUNCTION 0 I had to set wifi to 0 because an error popped up during compilation and now the printer prints normally [...]

i have the same problem with ROBIN E3P :(

alikPC, 2022-09-13:
hello please send corexy firmware info@terra-gidro.ru

padraardu123, 2024-02-27:
Hello, I made a printer with MKS Robin Nano v3.1 and everything works fine, but when I want to print, without doing anything, it very quickly says that the print is done. Please someone help me.
[Untitled.txt](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/14421946/Untitled.txt)