makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #124: Build error missing "qr_encode.h"

kalshashai, 2020-11-07:
### Description

During the build for Robin Nano 1.x following he instructions, the following error appear:
Marlin/src/lcd/extui/lib/mks_ui/draw_cloud_bind.cpp:36:10: fatal error: qr_encode.h: No such file or directory

### Steps to Reproduce

1. Install VS Code
2. Install Platform IO IDE extension from VS
3. Clone Mks-Robin-Nano-Marlin2.0-Firmware
4. Edit Configuation.h as follows:

       #define MOTHERBOARD BOARD_MKS_ROBIN_NANO

       #define TFT_LVGL_UI_FSMC

       //#define TFT_LVGL_UI_SPI

5. using Platform IO to build for env:mks_robin_nano35

**Expected behavior:** [What you expect to happen]
 successful build process

**Actual behavior:** [What actually happens]
 compilation terminated with the following errors:
      Marlin/src/lcd/extui/lib/mks_ui/draw_cloud_bind.cpp:36:10: fatal error: qr_encode.h: No such file or directory
      ........
     *** [.pio/build/mks_robin_nano35/src/src/lcd/extui/lib/mks_ui/draw_cloud_bind.cpp.o] Error 1
[image]
[Marlin.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5505422/Marlin.zip)

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.
[Marlin.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5505423/Marlin.zip)
[image]

kalshashai, 2020-11-07:
I think I just found the problem. _ do not know if it is going to work, still to test. I traced back the issue and found that in platformio.ini  Feature Dependencies section that the reference to qr_encode is "https://github.com/MKS-Sean/qr_encode/archive/master.zip". When I check the site, I found that the file names are in capital i.e "QR_Encode.h and QR_Encode.cpp" instead of "qr_encode". I was thinking that this should not be a problem. Just for testing, I edited  the calling file "draw_cloud_bind.cpp" and changed the include statement to "#include "QR_Encode.h". The I tried to compile and it compiled successfully.  What do you think is the issue? _

willianrampazzo, 2020-11-16:
I confirm the issue following the steps above to build the firmware.

PR #130 (2020-11-23): Avoid build break on missing qr_encode.h
### Requirements

* Filling out this template is required. Pull Requests without a clear description may be closed at the maintainers' discretion.

### Description

The build is breaking with "fatal error: qr_encode.h: No such file or directory" due to an incorrect file case. The correct file name listed at https://github.com/MKS-Sean/qr_encode/blob/master is `QR_Encode.h`.

I'm proposing the code change here, but it is also possible to change the file case at https://github.com/MKS-Sean/qr_encode/ .

### Benefits

Corrects the build.

### Configuration [... cut]
commit 4919fc55 (2020-11-23): Avoid build break on missing qr_encode.h

Build is breaking with "fatal error: qr_encode.h: No such file or
directory" due to an incorrect file case. The correct file name
listed at https://github.com/MKS-Sean/qr_encode/blob/master is
QR_Encode.h.

Fixes #124
closed 2020-11-22 by kalshashai