makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #125: Autohome after G29

Malderin, 2020-11-08:
### Description

Autohome (G28) always occurs after command G29 (auto level sensor required)

### Steps to Reproduce

Method 1 - from the printer menu (tools / autolevel)
Method 2 - send G29 through any slicer like cura

#### Additional Information

[conf.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5507041/conf.zip)

Oranginam, 2020-11-20:
Hi, in configuration.h there is :
/**
 * Commands to execute at the end of G29 probing.
 * Useful to retract or move the Z probe out of the way.
 */
#define Z_PROBE_END_SCRIPT "G28\nM500"

Modify gcode or comment the line and compile.

Malderin, 2020-11-20:
> * Z_PROBE_END_SCRIPT

you are right, thanks!

closed 2020-11-20 by Malderin