makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #127: Only Servo0 works

macace2000, 2020-11-12:
There is a problem, if you want to use more than one servo.

I want to use the switching nozzle feature and the BL-Touch.
If I disable BL-Touch and use Servo0 for the switching Nozzle (Pins.h:  Servo0 =PE5)
It works and the connected Servo move correct.

When I change the Switching nozzle to Servo1 (Pins.h:  Servo0 =PA8 / Servo1=PE5),
The BL-Touch works, but the Switching Nozzle not.

In the Configuration.h the Numbers of Servos is set to 2 (#define NUM_SERVOS 2)  

I have tried serveral Versions of Marlin, include the actual 2.0.7.2 form the main
Marlin Site, the Problem is in every Version.

mks-viva, 2020-11-18:
Hi! First of all, is this function achievable, or is it already achieved by users?

macace2000, 2020-11-20:
Hello,

I have discuss this also in the main Marlin branch.
See the solution at this issue:

https://github.com/MarlinFirmware/Marlin/issues/20119