makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #128: Dual Z stepper

UnleashSpirit, 2020-11-14:
### Description

I have a problem with double z stepper when Z homing.
I already set
#define NUM_Z_STEPPER_DRIVERS 2

When homing for Z, axis go up then down but about 3 sec later while going down, on stepper (the one on E1) goes up.
I took a video of it if it can help : https://youtu.be/KZiF46P9mJY

I don't had this problem when using Mks_Nano35_V2.0.3.2 firmware

#### Additional Information
board : MKS Robin nano v1.2
I use a BLTouch

mks-viva, 2020-11-18:
Hi! I has look your video and i think have two ways
1. You can exchange E1 motor 1A，1B position. Realize the change direction
2. You need check E1 motor direction settins in configuration.h file, it is same as Z axis

UnleashSpirit, 2020-11-18:
Hi,
I try
```c
#define INVERT_Z_DIR     true
#define INVERT_E1_DIR    true
```
No change

I try 
```c
#define E1_DIR_PIN    PB1
```
instead of PA1 in pins_MKS_ROBIN_NANO.h
It's worst, when down, motor don't even start in the same way than Z but directly the opposite. But strange, going up still OK

UnleashSpirit, 2020-12-07:
Hi, 
I don't know what changed but I pull last version just now and it works !

closed 2020-12-07 by UnleashSpirit