makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #129: Flsun Q5 - Robin Nano v1.2

eduwinner-1, 2020-11-14:
Hi, I compiled and installed Mks-Robin-Nano-Marlin2.0-Firmware on a FLSUN Q5 but icons are not shown correctly, seems like screen resolution is not right.

I tried to change some parameters in Configuration.h with no success

As per documentation, the config is:

#define TFT_LVGL_UI_FSMC  // Robin nano v1.2 uses FSMC
//#define TFT_LVGL_UI_SPI   // Robin nano v2.0 uses SPI

I tried to uncomment other TFT "define" but I can't compile the .bin

I attach a screenshot.

Does someone isntalled Marlin 2.0 in a Flsun Q5 sucessfully?
[image]

eduwinner-1, 2020-11-18:
any idea how to fix the resolution for TFT24 (I guess)?

mks-viva, 2020-11-18:
@eduwinner-1 
Hi! MKS Robin TFT24 have not MKS UI, so, you only use marlin color UI.

eduwinner-1, 2020-11-21:
I think the Q5 screen is TFT28, not TFT24.

Can we instal MKS UI on TFT28 display. I tried to compile marlin2.0 with the Robin Nano v1.2, I copied assets folder and Robin_nano35.bin (renamed to Robin_nano.bin) to the SD root folder. Then I turn on the printer and it update the firmware but I can't get the icons fixed...

mks-viva, 2020-11-23:
Only supported MKS Robin TFT35, because 2.4/2.8/3.2 have not support.

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva