makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #132: Question and request for help

Papapayana, 2020-11-19:
Hello im running the MKS Robin Nano v2 with a TS35 V2.0 and im wondering if theres a way to flip the screen orientation and if there is how do i do it?

themacboy, 2020-11-20:
Flip or rotate to vertical orientation?

Papapayana, 2020-11-20:
flip it so that it would be upside down when looking at it normally

themacboy, 2020-11-20:
duplicate #90

mks-viva, 2020-11-23:
@Papapayana 
Hi! You can setting it in configuration_adv.h file, enable #define GRAPHICAL_TFT_ROTATE_180
[image]

themacboy, 2020-11-23:
Is there a way to rotate? Set vertical screen?

Missatge de mks-viva <notifications@github.com> del dia dl., 23 de nov.
2020 a les 2:26:

mks-viva, 2020-11-23:
> Is there a way to rotate? Set vertical screen?

Not for the time being, because the size of the image is completely different, we will consider this feature.