makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #133: Printing end

WashingtonJunior, 2020-11-25:
Using TwoTrees Bluer Nano 1.2, if you leave another screen open (not the status screen) the current printing job won't "finish" properly. Leaving the screen open, when you return it says 100% complete on LCD, after actually finishing. In this case, you need to press the stop button to return to main menu. This also affects the wifi plugin, as Cura thinks that the printing is still going and didn't finish yet.

mks-viva, 2020-11-25:
Hi! Thank you for your feedback, I will arrange testers to reproduce this problem.

mks-viva, 2020-11-25:
This does not affect, because after Cure slices, the entire gcode file is transferred to the motherboard TF card through WIIF, and then print.

WashingtonJunior, 2020-11-25:
In Cura, if you switch to MONITOR mode (the one after PREVIEW), using the WIFI module, it shows the ongoing print or it will let you adjust temperatures, home/move axis, send custom G-Code, etc., the same way as if your printer is connected through USB. In this case, if the printing doesn't "finish", all these options are disabled and you can't send a new G-Code. When it finishes properly, it's all good (Monitor options are enabled) and you can send a new file as usual.

WashingtonJunior, 2020-11-25:
By the way, it shows 0% and not 100%, I made a video showing the issue: https://youtu.be/IVQmZ64WpYA

X-Ryl669, 2020-11-25:
Thanks for this report, it's linked with #126 it seems, and probably fixing the former will fix the latter.

WashingtonJunior, 2020-11-25:
Yes, just recently I noticed that the former was caused by not staying in the status screen. Thank you!

mks-viva, 2020-11-26:
@WashingtonJunior 
Ok! Thank you for your feedback, I will communicate this issue with our engineer.

MKS-Sean, 2021-02-21:
@WashingtonJunior @X-Ryl669 I have fixed this issue in the latest submission.

closed 2021-02-21 by MKS-Sean