makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #134: Merge with latest Marlin release

X-Ryl669, 2020-11-26:
Currently, the difference with official Marlin aren't that big (thanks!). So I'm wondering how hard it would be to maintain a branch with only the pending changes with Marlin's version ? 
I'm thinking of the few changes you have (WIFI, translation, etc).

I've tried to merge latest Marlin with your version, but since Marlin is doing a HUGE lot of contributions, it's hard to do (I haven't finished yet). I'm wondering if the opposite wouldn't be better that is:

1. Make your changes as a branch and rebase over Marlin's firmware regularly (you have less work than Marlin)
2. Report issues to Marlin and use their fixes (for example, there are many opened issue like the non working babystepping that are Marlin's fault and have been fixed there but not here yet)

This would save you a lot of maintenance work to avoid dealing with issues not from your code.

X-Ryl669, 2020-11-27:
I've started this work [here](https://github.com/X-Ryl669/Marlin/tree/MKSBranch).
For now, I've applied all the recent commit since last Marlin 2.0.6 version onto Marlin's repository.
I'm currently rebasing those on the latest Marlin version. If it's too much work, I'll probably squash them to a single commit (or very few commits).

X-Ryl669, 2020-12-02:
Ok, I've finally applied all changes from this repository on latest Marlin [here](https://github.com/X-Ryl669/Marlin/tree/bigChange)
That was a very painful and long work. 

Can any of Makerbase (@MKS-Sean, @mks-viva, @Malderin, @makerbase-mks) checks that nothing is missing or break in your tests ?

My idea would be to maintain the latest changes from Mks on top of official Marlin so we don't lag and have to fix issues that are already fixed in Marlin. Ideally, if done correctly, one could drop this repository completely since it's very difficult to maintain as it is with so many divergence with official code.

WashingtonJunior, 2020-12-04:
Great idea, this weekend I'll try to use your fork to see if it solves some issues. Thanks for sharing!

X-Ryl669, 2020-12-09:
I've finished fixing all the building bugs for this fork last night. If you have a Sapphire Plus, you'll even find a pre-build firmware for testing in that repository. I'll then sort out the bug I might have introduced by running few prints in the next days.

WashingtonJunior, 2020-12-12:
Does it works on your Sapphire Plus? I have a Bluer and tried both 'bigChange' and 'Sapphire' branches, but the printer keeps restarting (without SD Card inserted) or restart randomly while using the menu (with SD Card inserted). Do you have any idea what could be?
For 'bigChange' I've updated Configuration.h with my settings and for 'Sapphire' I've just uncommented "#define BLUER" and commented the original "#define SAPPHIRE_PLUS".

Thanks!

X-Ryl669, 2020-12-12:
I'm experiencing some issues for now (it restarts when it locks due to watchdog, and it locks when listing the files on the SD card for me). The strange thing I've spotted so far is, upon boot, it says "freeMemory: -27" which I think is the main cause of the bugs, somehow, we might be consuming too much memory. While merging, I've left a large buffer for the bmp_picture that was reduced in this repository (not sure why they did that, but now I might understand) (14 instead of 17). I'll put it back to the value in this repository and let you know. 

Do you have the Configuration (.h and adv.h) you are using ? I'll make a Blue branch with your files.

WashingtonJunior, 2020-12-12:
Sure, as attachments you'll find both attempts (using bigChange and Sapphire branches). Thank you!

[ConfigBluer_SapphireBranch.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5683635/ConfigBluer_SapphireBranch.zip)
[ConfigBluer_bigChange.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5683636/ConfigBluer_bigChange.zip)

X-Ryl669, 2020-12-13:
Ok, this was indeed the cause of my issues. You can retry now with whatever branch. I haven't fixed all the bugs yet, but those that remains are in MKS's branch also, so it's not those induced by my merge.

MKS-Sean, 2020-12-14:
@X-Ryl669 @WashingtonJunior 
We also submitted a version to the marlin official, and we also want to merge it into the marlin official as soon as possible. Eventually, the official version of marlin will be the main one, but we have encountered some problems and are still waiting to be resolved.

X-Ryl669, 2020-12-14:
Are you talking about this [PR](https://github.com/MarlinFirmware/Marlin/pull/20191) ?
In my branch, I'm tracking the latest Marlin version (that is, **not** the `bugfix-2.0.x` branch of Marlin which is outdated). It's a lot more work than trying to include WIFI into an old branch that'll die sometime in the future, if I understand Marlin's way of releasing correctly. 

My work should solve future update of Marlin since it's porting the **current** MKS code into **current** Marlin master `2.0.x` branch.

If you want to compare your work with mine, please use this [link](https://github.com/MarlinFirmware/Marlin/compare/1c7a4f3c58cc4a85ea19d83dba208f1bddf27fd5...X-Ryl669:bigChange)