makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #135: PlatformIO loads all boards.

anteyru, 2020-11-26:
Downloaded from your site Marlin added to the VS code downloads tens of thousands of files for different platforms and boards. The computer has been trying to download all this for more than an hour and it is not possible to compile the firmware.
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = mks_robin_nano35
include_dir  = Marlin
-----------------------------------------------
.pio\libdeps\
STM32F103VE_longer\
STM32F103VE_GTM32\
STM32F103RE_btt_USB\
STM32F103RE_btt\
STM32F103RE\
STM32F103RC_meeb\
STM32F103RC_fysetc\
STM32F103RC_btt_USB\
STM32F103RC_btt_512K_USB\
STM32F103RC_btt_512K\
STM32F103RC_btt\
STM32F103RC\
STM32F7\
STM32F4\
sanguino1284p\
sanguino644p\
SAMD51_grandcentral_m4\
rambo\
mks_robin_nano35\
mks_robin_mini\
mks_robin\
MightyBoard2560\
MightyBoard1280\
melzi_optimized\
melzi_optiboot\
melzi\
mega2560\
mega1280\
LPC1769\
LPC1768\
FYSETC_F6_13\
DUE_USB\
DUE_debug\
DUE_archim_debug\
DUE_archim\
DUE\
at90usb1286_dfu\
at90usb1286_cdc\
ARMED\
mks-assets.zip
------------------------
.pio\build\
mks_robin_stm32\
melzi\
project.checksum

X-Ryl669, 2020-11-27:
That's how platformio works. You only need to do it once.

mks-viva, 2020-12-11:
You can refer to Nano Wiki
MKS Robin Nano V1.x: https://github.com/makerbase-mks/MKS-Robin-Nano-V1.X/wiki
MKS Robin Nano V2.x: https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki

mks-viva, 2020-12-11:
I will close this issues temporarily

closed 2020-12-11 by mks-viva