makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #137: Pause and resume

WashingtonJunior, 2020-12-03:
The extruder won't move forward, it just retracts, after hit pause and then resume. It already happened twice here with me, while doing a multi-color print. But, it won't occur if you do the pause and resume on the first layer (yes, I tested) at least. It only happens after some time printing. I made a video showing, if it's needed.

Thanks!

mks-viva, 2020-12-04:
Hi! You can provide it to me.

WashingtonJunior, 2020-12-04:
This one is after several layers (you'll see that the extruder won't move forward, only retracts): https://youtu.be/E5g7o1HHf7Y
And this one is on first layer (when it continues as usual): https://youtu.be/gW3MP46AHrI

mks-viva, 2020-12-07:
Very similar to this issues: https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/133

WashingtonJunior, 2020-12-07:
> Very similar to this issues: #133

No, in this case when we pause and resume it continues, but without extruding. The other issue is that the printing does end, but it shows on display like it's still going on.

mks-L, 2020-12-09:
Can you provide the Configuration Files to me?

WashingtonJunior, 2020-12-09:
> Can you provide the Configuration Files to me?

Sure, here they are. Thanks!

[Configuration-files.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5663008/Configuration-files.zip)

mks-L, 2020-12-09:
May I ask which version of the file you used? This problem does not occur when I use the new file.

mks-L, 2020-12-09:
You can use the latest files to update and see if the same problem occurs.

misiek271, 2020-12-20:
Iv got robin v2 and the same is in my printer

n9jcv, 2022-03-07:
Yes, I have this same issue,  Robin Nano V2.  Pause, then resume and extruder no longer extrudes.

n9jcv, 2022-03-11:
I tested with Marlin 2.0.9.2 
COMMIT 4E2443A476
And the bug is still there.  Can someone please fix this ?
config files attached
[configs.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/8228679/configs.zip)

rjhubner, 2022-05-11:
Same is happening to me on pause or even on end of filament that calls the pause routine. I have tested a couple of Robin Nano V2 and same problem. 
My Marlin version 2.00.06

freeman02, 2022-06-02:
Yes, I have this same issue, Robin Nano V1.2. Pause, then resume and extruder invert
Marlin version 2.0.9.2