makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #139: S3D : Firmware unresponsive with Robin Nano

Elzariant, 2020-12-05:
Printer : TEVO Little Monster - Delta Printer
Board : MKS Robin Nano v2 + TMC2225 Drivers + TS35 screen
Slicer : Simplify3D

Hello, I'm trying to update my TEVO Little monster using MKS Robin V2 + TS35 screen. I manage to update Marlin 2.0 configuration. I cannot print via SD card : the print starts as both Bed and Hotend heat but then nothing happens. That issue will be for later. With Simplify3D, everything seem fine except the print randomly stops, raising "WARNING: Firmware unresponsive.  Attempting to force continue..." in S3D. I tried with several USB cables that work perfectly on my other printer but no luck. Has anyone also faced that issue ? 

Note : here are other issues I'm facing
UART and "Manual" modes : Prints from SD Card : Not working
UART mode : works partially as printer movements look weird. Sometimes the hotend draws big triangles outside the print, unknown command error are raised in the slicer, and stepper torque is not at expected level whatever value of current.

mks-viva, 2020-12-07:
Hi! I want to know, can the machine work normally before printing? If it works, then, if you use Cura to slice and print, will it work?

Elzariant, 2020-12-07:
Hello,
Before I print, everything seems fine. Even the start of the print is OK during few layers until the problem occurs. I tried to print via Cura too as I suspected an issue with GCode generated by S3D but unfortunately that does not solve the issue, the print stops suddently. As I'm not a Cura expert nor user, I don't know where I can see the messages coming coming from the printer.

I tried to print the exact model on another printer on which I put an MKS SGEN_L v2 with TMC2225 and I don't face any issue. 

New test I'll do tomorrow is to use the official latest version of Marlin (2.0.72) and update it for my Delta printer (add QR_Encode in the mks_ui folder and adapt configuration)

Elzariant, 2020-12-08:
News : I manage to have latest Marlin version (2.0.72) working on the printer ... or so I thought. New version, new issue (sic!). With both Cura and Simplify3D, prints are launched properly. But Extruder only work during few seconds before its stepper stops suddently and then refuse to work anymore - as if it was disable during the print. 

If I reset completely the printer, heat the nozzle to 210°C and launch extrude commands, it works fine. But during prints, it stops. What could stop the extruder during the print except a too-low temperature ?

(Honestly, that's the first time I face so much issue with a board from MKS. I got almost all models (Base, Gen, SBase, GEN_L, SGEN_L) in almost all revisions and they always work properly without any trick).

mks-viva, 2020-12-09:
You use MKS UI? If you use marlin color UI or marlin 12864 UI, whether the problem does not exist?

Elzariant, 2020-12-09:
Hello,
I normally  use the TS35 color UI. But in my last test, I completely disconnected the UI cables removed) and drove the printer only using USB connection. And once again, the extruder stopped after few seconds of print, either with S3D and Cura.

Elzariant, 2020-12-10:
I discovered that the "extrusion stop" issue was coming from LinearAdvance feature. Strange that I did not encounter it previously as it has always been activated. Nevertheless, once the feature deactivated, prints are back with the previous issues.

mks-viva, 2020-12-11:
It looks more like firmware.

Elzariant, 2020-12-11:
New tests. TMC2225 drivers are an update to 2208 one so I looked after issues with these drivers and applied all known corrections.

- I disabled stealthchop : NOK
- I disabled Linear_Advance : NOK
- #define MINIMUM_STEPPER_PULSE 1 : NOK
- #define MAXIMUM_STEPPER_RATE 5000000 : NOK
- #define SQUARE_WAVE_STEPPING //added : NOK
- Reduced speed from 120mm/sec to 80mm/sec : seemed better but extruder stoped later in the print : NOK
- Reduced jerk : NOK

--> I noticed that stepper stops after a retraction. It seems it does not like rapid change of direction

Finally, I surrendered with TMC2225 on Extruder and switched it to an old TMC2100 I got. I adapted the steps-per-units and prints are now OK. I wonder why this "disabling-stepper" issue/feature only affects the extruder and not the other steppers. That's a delta-printer so all steppers are moving fast constantly.

Elzariant, 2020-12-13:
I'll close the initial issue and open a new one concerning the extruder that stops suddently after a retractation. The problem remains with TMC2100 driver.

closed 2020-12-13 by Elzariant