makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #140: MKS PWC not working

I-esus, 2020-12-06:
Hello!

I'm running the MKS Robin Nano v1.2 + MKS Robin TFT 35 v1.0 + MKS PWC v3.0. Worked fine. 

Now I had to change the firmware due to a thermistor change and since then it is not working. I put the SD, the firmware updates but does not turn on. I removed the PWC, just Robin Nano + TFT 35 and it works.

What I should modify on Marlin to work with the PWC?

mks-viva, 2020-12-07:
Hi! Haha, you need to keep pressing the PW_OFF switch when you need to update the firmware.

I-esus, 2020-12-07:
Not working.
It needs to uncomment "#define PSU_CONTROL" in configuration.h?

mks-viva, 2020-12-08:
That is necessary, you need enable PSU_CONTROL

I-esus, 2020-12-08:
Ok, enabled PSU_CONTROL and the pins (attached the files).

It does not compile due to this error:
> In file included from Marlin/src/HAL/STM32F1/../../inc/MarlinConfig.h:41:0,
                 from Marlin/src/HAL/STM32F1/HAL_SPI.cpp:35:
Marlin/src/HAL/STM32F1/../../inc/SanityCheck.h:2933:6: error: #error "PSU_CONTROL requires PS_ON_PIN."
     #error "PSU_CONTROL requires PS_ON_PIN."

[Configuration.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5658115/Configuration.zip)
[pins_MKS_ROBIN_NANO.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5658118/pins_MKS_ROBIN_NANO.zip)

mks-viva, 2020-12-09:
I think you need to provide me with the entire firmware so I can check it

I-esus, 2020-12-09:
Here it is: [https://we.tl/t-a6z5oNQfEv](url)

mks-viva, 2020-12-10:
I am very sorry. I corrected my mistake. If you use MKS UI, you only need to set the pin pins in the pins_MKS_ROBIN_NANO_V2.h file. No other settings are required. For this, I will complete the MKS Robin Nano V2 Wiki.

UrknallBigBang, 2020-12-12:
Yes you don't need to define PSU_CONTROL you have to define in pins_MKS_ROBIN_NANO_V2.h like this
#if HAS_TFT_LVGL_UI
  #define MKSPWC    //uncomment this so it is active
  #ifdef MKSPWC

    #define SUICIDE_PIN                                PB2   // Enable MKSPWC SUICIDE PIN
    #define SUICIDE_PIN_INVERTING             false // Enable MKSPWC PIN STATE
    #define KILL_PIN                                       PA2   // Enable MKSPWC DET PIN
    #define KILL_PIN_STATE                            HIGH // Enable MKSPWC PIN STATE
  #endif

And you have to hold the button when you do the update. When update is done then switch on the printer and hold button until MKS Screen appeares. When you hold too long then Error Printer hold appears. Do it once more an let the button go when MKS Screen appeares. All done have fun.

mks-viva, 2020-12-30:
@I-esus Can your MKS PWC work?

I-esus, 2020-12-31:
Sorry but I couldn't try it yet. I will let you know when I do it.

closed 2021-02-21 by MKS-Sean