makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #141: Baby stepping is not working

artdezigner, 2020-12-06:
Hi, MKS Robin Nano v2, your version Marlin, mks graphical menu, baby stepping is on, menu is on, but buttoms is unactive, please help. Marlin code:
#define BABYSTEPPING ////
#if ENABLED(BABYSTEPPING)
  //#define INTEGRATED_BABYSTEPPING         // EXPERIMENTAL integration of babystepping into the Stepper ISR
  //#define BABYSTEP_WITHOUT_HOMING //
  //#define BABYSTEP_XY                     // Also enable X/Y Babystepping. Not supported on DELTA!
  #define BABYSTEP_INVERT_Z false           // Change if Z babysteps should go the other way
  //#define BABYSTEP_MILLIMETER_UNITS       // Specify BABYSTEP_MULTIPLICATOR_(XY|Z) in mm instead of micro-steps
  #define BABYSTEP_MULTIPLICATOR_Z  1       // (steps or mm) Steps or millimeter distance for each Z babystep
  #define BABYSTEP_MULTIPLICATOR_XY 1       // (steps or mm) Steps or millimeter distance for each XY babystep

  //#define DOUBLECLICK_FOR_Z_BABYSTEPPING  // Double-click on the Status Screen for Z Babystepping.
  #if ENABLED(DOUBLECLICK_FOR_Z_BABYSTEPPING)
    #define DOUBLECLICK_MAX_INTERVAL 1250   // Maximum interval between clicks, in milliseconds.
                                            // Note: Extra time may be added to mitigate controller latency.
    //#define BABYSTEP_ALWAYS_AVAILABLE     // Allow babystepping at all times (not just during movement).
    //#define MOVE_Z_WHEN_IDLE              // Jump to the move Z menu on doubleclick when printer is idle.
    #if ENABLED(MOVE_Z_WHEN_IDLE)
      #define MOVE_Z_IDLE_MULTIPLICATOR 1   // Multiply 1mm by this factor for the move step size.
    #endif
  #endif

  //#define BABYSTEP_DISPLAY_TOTAL          // Display total babysteps since last G28

  //#define BABYSTEP_ZPROBE_OFFSET          // Combine M851 Z and Babystepping
  #if ENABLED(BABYSTEP_ZPROBE_OFFSET)
    //#define BABYSTEP_HOTEND_Z_OFFSET      // For multiple hotends, babystep relative Z offsets
    //#define BABYSTEP_ZPROBE_GFX_OVERLAY   // Enable graphical overlay on Z-offset editor
  #endif
#endif

WashingtonJunior, 2020-12-06:
I don't know if helps, but I've uncommented the same lines as you (but including BABYSTEP_DISPLAY_TOTAL) and it works for Nano 1.2, although it doesn't display the amount.

mks-viva, 2020-12-07:
@artdezigner 
Hi!
The premise of baby stepping function
1. It must have automatic leveling function
2. Baby stepping needs to be adjusted during printing

artdezigner, 2020-12-07:
> @artdezigner
> Hi!
> The premise of baby stepping function
> 
> 1. It must have automatic leveling function
> 2. Baby stepping needs to be adjusted during printing
Hi, Yes, that's right. I have an inductive z-axis sensor installed, and auto-level is enabled. When printing, there is a babystepping menu, but the buttons in it do nothing when clicked

artdezigner, 2020-12-08:
What should I do? Bought your Board instead of v1 + TFT 32 only because of Linear advance and Babystepping... and here are the problems again

WashingtonJunior, 2020-12-08:
I don't have an automatic leveling sensor and Babystepping works here (it doesn't display the amount, but it raises Z when I tap the button).

mks-viva, 2020-12-09:
@WashingtonJunior 
Hi! Can you share your settings and let more people use it?

WashingtonJunior, 2020-12-09:
> @WashingtonJunior
> Hi! Can you share your settings and let more people use it?

Sure! My configuration files follows as an attachment. They are properly configured for a Two Trees Bluer v1.

[Configuration-files.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5666868/Configuration-files.zip)

artdezigner, 2020-12-09:
it worked when I enabled this #define BABYSTEP_PROBE_OFFSET

mks-viva, 2020-12-10:
@WashingtonJunior
Thank you! @artdezigner looks like everything is perfect

closed 2021-01-28 by MKS-Sean