makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #142: В стоке не работает обдув радиатора хотенда (решение)

artdezigner, 2020-12-06:
Если не работает обдув радиатора, то:
1. В файле Configure.adv.h нужно прописать
#define E0_AUTO_FAN_PIN FAN1_PIN
2. В файле pins_MKS_robin_nano_v2.h прописать вот так:
//
// Heaters / Fans
//
//#define HEATER_0_PIN                        PC3   // HEATER1
//#define HEATER_1_PIN                        PB0   // HEATER2
//#define HEATER_BED_PIN                      PA0   // HOT BED

//#define FAN_PIN                             PB1   // FAN

////
#define HEATER_0_PIN       PC3
#if HOTENDS == 1
  #define FAN1_PIN         PB0
#else
  #define HEATER_1_PIN     PB0
#endif
#define FAN_PIN            PB1
#define HEATER_BED_PIN     PA0
////

Ранее такой же косяк был и на платах v1.X, но потом в пинах дописали, а про плату v2.X забыли...

mks-viva, 2020-12-07:
Hi! Thanks for your feedback, we will test and update it.

MKS-Sean, 2021-02-21:
@artdezigner MKS Robin nano V3 has FAN1_PIN, I am worried that it will be more troublesome to use

Yujev, 2023-05-29:
Спасибо!