makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #143: MKS Robin Nano TFT3.5 showing Multi-color Stripes

selvakumaran2016, 2020-12-07:
Hi,

We are using Robin Nano V2.0 with TFT 3.5 in Marlin mode. (without knob). The screen gets disturbed often. It is changing the colors and going with multi-color stripes.
[image]

Please suggest some solution for this.

mks-viva, 2020-12-08:
https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/issues/23
I need you to do a few tests
1. Take out the MKS TS35 V2 display screen and not install it on the machine. Will there still be problems with the Chinese language of your picture?
2. I want to see if the connection line between your screen and the motherboard has passed through interference sources such as power supply

closed 2021-02-21 by MKS-Sean

fatih5228, 2023-12-22:
For display problem, connect 11mm Ferrite Core to EXT1 output.
[https://www.amazon.com/Ferrite-Clamp-Cores-Round-Filter/dp/B005T8UQAM/ref=sr_1_14?crid=3TUDQXEN3TLCO&keywords=Ferrite+Core&qid=1703252127&sprefix=ferrite+core%2Caps%2C211&sr=8-14](url)
or
[aliexpress.com/i/4000803662446.html?gatewayAdapt=glo2tur](url)