makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #145: [Robin Nano 2.0] Printing stops shortly when touching the MKS TS35 display

ikkez, 2020-12-13:
When the printer is printing and I touch some menu items to either change temperature or whatever, the printer stops for a short duration while it's loading the next menu to display.. it's not very short but 0.5 - 1 second and will leave a mark in the print. Why is that so? Do I have to many features enabled in the firmware or is the Nano v2 just so slow that it cannot refresh the display while printing fast enough?

**Expected behavior:** Immediatly display the menu and do not stop printing while loading menu

**Actual behavior:** Printer freezes for a secound while it's loading the menu item to show on the MKS TS35 display

Elzariant, 2020-12-13:
Hello, 

Same issue for me. Touching the TS35 screen during a print makes the print freezes a sec.

mks-viva, 2020-12-14:
@ikkez @Elzariant 
Hi! Can you provide related videos when printing is performed.

ikkez, 2020-12-14:
please see this video here
[IMG_0086.mp4.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5690248/IMG_0086.mp4.zip)

mks-viva, 2020-12-15:
Should be caused by the execution of a short path, i will feed it back to our engineers.

Elzariant, 2020-12-15:
Here is my video. Exactly the same issue as ikkez : 

[Robin Nano v2 Issue - Video - Google Drive](https://drive.google.com/file/d/162Eicqcb72Ywfo9dHEd4pB6ZwZQcSlmv/view?usp=sharing)

Elzariant, 2020-12-29:
Hello, any feedback regarding that issue ?

mks-viva, 2020-12-30:
Use Mks-Robin-Nano-Marlin2.0-Firmware？Is it the latest version of the firmware？

Elzariant, 2021-01-04:
Have you changed anything in that firmware ? My version is 20 days old.

pistachin2, 2021-01-28:
With the version to day of today (28-01) it continues happening.

Elzariant, 2021-02-11:
Same issue here. Nothing changed.

ikkez, 2021-02-13:
I have it tested today with latest firmware and the issue is still present. To me it looks like the V2 chip is to weak to handle printing and rendering the graphical menu at the same time. I've ordered the new nano v3 board and will update here if that makes anything different

ikkez, 2021-12-28:
So.. one year later, I've finally upgraded to nano v3 board, compiled a fresh updated firmware and the problem is still there.
The problem seems to be somewhere in the `TFT_LVGL_UI`. 
With the firmware compiled to use `TFT_COLOR_UI` instead, the problem does not happen... but then it's all ugly of course and some features are missing.

ikkez, 2022-02-04:
Is this an issue of the FTF35 firmware or the used UI driver in the firmware? Any idea?