makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #146: Extruding in Pause Menu is in Relative Mode

ikkez, 2020-12-13:
When I pause the print and go to extrude menu and extrude 50mm of filament and then hit resume, the printer very fast retracts that 50mm again. 

### Steps to Reproduce

1. Start a print
2. Pause the print
3. Goto extrude menu
4. Extrude 50mm
5. go back and resume print

**Expected behavior:** Extruder behaves like normal

**Actual behavior:** Extruder retracts the same amount of filament I manually extruded in the menu. This is probably an indicator that the extrude command was made in relative mode and not absolute.

mks-viva, 2020-12-14:
Hi! I want to know:
1. What motherboard and screen are you using？
2. Which firmware do you use?

ikkez, 2020-12-14:
Hi, I'm using:

Hardware: MKS Robin Nano v2.0 with MKS TS35 TFT screen.
Firmware Version: Marlin-bugfix2.0.x-MKS-2.1.1

MKS-Sean, 2021-01-09:
@ikkez Hi，I fixed this problem and it was normal after the test. If you have tested a new version, please let me know the test result

closed 2021-01-18 by MKS-Sean

ikkez, 2021-02-14:
Hi @MKS-Sean ... I have tested today again but the problem is still present. I also cannot see which change you made in your commit https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/commit/90780f8a5d79c6c54a0f1a5f354d5f6a5cd14085 that was related to this issue. Can you please clearify that, so I can make some own tests? It not very useful to put 4000 changes into a single commit that should look like to fix my reported issue.