makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #147: Tft Rotation

telemac73, 2020-12-15:
I have update from nano 1.2 to 2 and to put screen on my printer i have to rotate it.

By default i see was rotated in marlin :#define TFT_ROTATION TFT_ROTATE_180 with that touch screen work well.
But to have it in the right sense on my printer i have to modify to :#define TFT_ROTATION TFT_NO_ROTATION and after that all touch screen button are wrong (not proprely rotate)

Any idea how to fix that ?
Thanks

mks-viva, 2020-12-16:
Hi! You use Mks-Robin-Nano-Marlin2.0-Firmware?

telemac73, 2020-12-16:
You are right i take the wrong distrb , but i just take a try with the good one but  TFT_ROTATION TFT_ROTATE_180 dont work ?
Thank

mks-viva, 2020-12-18:
Ok! I will test it

mks-viva, 2021-01-07:
Hi! You can set it and test it temporarily like this:  
[image]

We will fix it

mks-viva, 2021-01-07:
I test it is ok.
[image]

mks-viva, 2021-01-07:
My test firmware bin file
[Robin_nano35.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5780158/Robin_nano35.zip)

telemac73, 2021-01-07:
Hello )
Thanks a lot that work fine now ))
Have a nice day

mks-viva, 2021-01-07:
Thank you for your support of MKS products. So, i will close this issues temporarily.

closed 2021-01-07 by mks-viva