makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #148: TFT43

tunico14, 2020-12-15:
I went to compile for TFT43 and I didn't find enough information for it to compile for this display. Already have support for TFT43? I could add support for it, as I want to update my two trees bluer plus. Thanks

mks-viva, 2020-12-16:
Hi! Use Mks-Robin-Nano-Marlin2.0-Firmware?

tunico14, 2020-12-16:
Yes, I would like to use the firmware marlin. Two Trees Bluer Plus comes with customized nano v1.0.3 software. How would I support this new version of marlin for my printer? It comes with a Robin Nano v1.2 and TFT43. Thanks

mks-viva, 2020-12-17:
Hi! Marlin firmware has not support Robin TFT43

tunico14, 2020-12-17:
Is it possible for you to add support for it in the near future?

mks-viva, 2020-12-18:
Yes, you can refer to here: https://github.com/MarlinFirmware/Marlin/pull/20334

rhapsodyv, 2020-12-29:
The marlin support for 4,3” screen was merged!

tunico14, 2021-01-03:
Is it possible to download the firmware already compiled for the TFT43, similar to the already available TFT35?

mks-viva, 2021-01-04:
@tunico14 This may take time.

closed 2021-09-02 by mks-viva