makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #149: System keep booting it self

froze1, 2020-12-26:
Hi like title said ,system booting it self when ı touch something.

here's the video about it:

[20201226 193448-1.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5743423/20201226.193448-1.zip)

mks-viva, 2020-12-29:
Hi! Which firmware are you using?

froze1, 2020-12-29:
Hi,

latest firmware(Mks-Robin-Nano-Marlin2.0-Firmware),just with some modification ı add.
So interesting because same firmware working well my firend bluer but not working  mine.(by the way we have same motherboard v1.2) 

[Bluer_source-20201226T194702Z-001.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5751196/Bluer_source-20201226T194702Z-001.zip)

mks-viva, 2020-12-30:
Can I understand it this way, if you do not modify the firmware, then it can work normally? You modified the firmware, it doesn't work?

froze1, 2020-12-30:
Yes ,you are correct normal firmware work normally . Like ı said at the beggining  for some users this modify firmware ı made  working well but for my system booting it self.

mks-viva, 2020-12-30:
Can you tell me which version you modified based on? You can try to use the latest version to modify according to your needs.

froze1, 2020-12-30:
(Marlin-bugfix2.0.x-MKS-2.1.1) my current firmware ,maybe ı can try old version for modification .

mks-viva, 2020-12-30:
Obviously, the failure occurred after the modification was made

froze1, 2020-12-30:
Ok,I will check again.Thanks

closed 2020-12-30 by froze1