makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #151: M600 on MKS Robin nano v1.2 with mks tft35  with mksPWC

thiodor93, 2021-01-06:
hi ! 
i 've  a ELF Creativity based on mks robin nano v1.2, display mks tft 35 v1.0 , mkspwc, i've filament runout detection ,
when active the voice #define FILAMENT_RUNOUT_SENSOR and FILAMENT_RUNOUT_SCRIPT "M600" , 
therefore ADVANCED_PAUSE_FEATURE  , i've this error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)." but stm32f1 does't support this, I have read in other posts similar to that which recommend creating another line with the m25 command, but the head stays on the print in this way, I don't understand if I'm wrong because I'm inexperienced or if it's a BUG!

mks-viva, 2021-01-07:
Hi! Like this https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/140
[image]

thiodor93, 2021-01-07:
> Hi! Like this #140
> [image]

thanks for the answer, but I have not had problems on the Pwc configuration, I am having problems with the filament runout sensor, since when I define it, I cannot park the extruder ...

mks-viva, 2021-01-07:
You does not define FILAMENT_RUNOUT_SENSOR, you only set it in pins_MKS_ROBIN_NANO.h file
[image]

thiodor93, 2021-01-07:
Ah .....ok thank you mr! But , how to set the nozzle parking?

mks-viva, 2021-01-07:
You can set it by Robin TFT35 display screen on line.
**Settings->Config->Advanced settings->Printing pause position settings** Set position

thiodor93, 2021-01-07:
Ah ok thank you mr. I correct the firmware and this week i try to install!

closed 2021-01-18 by MKS-Sean