makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #152: M25 or M600 using MKS UI ...

themacboy, 2021-01-10:
Good mornign trying to figure how to setup M25 nad M600 with TFT_LVGL_UI_SPI, but never success. 

Looks like it is only possible using Marlin UI.

Still no solution with mks_ui? I encourage you to find a way to run taht Gcodes.

Thx in advance !

mks-viva, 2021-01-11:
Hi! What motherboard and screen are you using?

themacboy, 2021-01-11:
Oh sorry, I forget to explain hehehe

Robin Nano v. 2
Ts35

mks-viva, 2021-01-11:
Haha! You can refer to here : https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/MKS_FES
[image]

themacboy, 2021-01-18:
Buit this is for filament sensor and filament change. But for example if I only want to use a M25, to pause the print move the nozzle to park zone, and then insert a nut inside the printed object. And after that continue the print.

How can I do that, G25 and park fucntion is not working for me. Can you test it? With Mks UI and marlin UI, thanks in advance !

mks-viva, 2021-01-19:
@themacboy 
Set the pause position, the same path setting：
Settings->Config->Advanced settings->Printing pause position settings Set position

themacboy, 2021-01-20:
> @themacboy
> Set the pause position, the same path setting：
> Settings->Config->Advanced settings->Printing pause position settings Set position

that obvious. But Z raise not working and more important after pause nozzle dint go to park zone.

that update will solve all this problems?
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/commit/8c610094b36810b9fe60894475384fcef7b2a260

Please mark this issue as solved if yes.

nando3100, 2024-06-15:
hola! yo tengo una MKS Robin Nano v3.1 y sin llegar a este foro hice lo mismo asigne el PIN PA4 en la linea 161 en el archivo pins_MKS_ROBIN_NAN_V3_coomon.h

154   #if HAS_TFT_LVGL_UI
155   #define MT_DET_1_PIN                      PA4   // MT_DET
156   #define MT_DET_2_PIN                      PE6
157   #define MT_DET_PIN_STATE                  LOW
158   #endif
159
160   #ifndef FIL_RUNOUT_PIN
161   #define FIL_RUNOUT_PIN                      PA4
162   #endif
163   #ifndef FIL_RUNOUT2_PIN
 164  #define FIL_RUNOUT2_PIN           MT_DET_2_PIN
165   #endif

y pude activar ->
En configuration.h

Habilitar #define FILAMENT_RUNOUT_SENSOR
Habilitar #define NOZZLE_PARK_FEATURE
En pins_MKS_ROBIN_NANO_V2.h

Añadir #define FIL_RUNOUT_PIN PA4 //Puede cambiar a otro pin
En confiuration_adv.h

Habilitar #define ADVANCED_PAUSE_FEATURE

incluido el M600
y funciona cualquiera de estas que actives
//#define TFT_CLASSIC_UI
//#define TFT_COLOR_UI
#define TFT_LVGL_UI

Saludos