makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #153: MKS Robin Nano v.2 USB issue ...

themacboy, 2021-01-16:
Good morning,

after last update a lot of things has changed with the introduction of v.3 parameters.

My problem is that now I have a lot of issues while compiling and nothing works.

For example:

1. you has added in the pin file of robin nano v.2 the next line:

#define BOARD_NO_NATIVE_USB

and now I receive that error:
Marlin\src\HAL\STM32F1\../../inc/SanityCheck.h:2387:6: error: #error "SERIAL_PORT is set to -1, but the MOTHERBOARD has no native USB support. Set SERIAL_PORT to a valid value for your board."

then I can use my usb like usual? or I must do any change? What changes? Please comunity help me.

Thanks in advance !

MKS-Sean, 2021-01-16:
Hello, I kept the parameters of nano_v2 in the latest submission, you can try it, my test is normal. If you are in trouble due to the upgrade of marlin, I suggest you use the original version first, but I have been constantly testing and improving the current version.

themacboy, 2021-01-17:
Please check 

#define BOARD_NO_NATIVE_USB

in the pins_MKS_ROBIN_NANO_V2.h file:

[image]

Thast really correct in that board? Thta is the line that generates the problem.

Not sure if you understand me.

Why you set there is no native usb in the mks robin nano v2?

MKS-Sean, 2021-01-18:
Can you provide configuration.h file and configuration_adv.h file？

hakimio, 2021-01-18:
@themacboy it's supposed to be like that:
https://github.com/MarlinFirmware/Marlin/issues/20221#issuecomment-732399620
https://github.com/MarlinFirmware/Marlin/pull/20047

themacboy, 2021-01-20:
My robin nano v2 configuration.h file:
[image]

My robin nano v2 pin file:
[image]

My error during compiling:

[image]

themacboy, 2021-01-20:
Ok, then I understand that usb support is really not native, and we must stop using -1 as port. And select any other?

hakimio, 2021-01-20:
Like Jason Smith wrote already, move the value from `SERIAL_PORT_2` to `SERIAL_PORT` and comment out `SERIAL_PORT_2`.

MKS-Sean, 2021-01-23:
@themacboy Hello, if you use the wifi function:
#define SERIAL_PORT 3
//#define SERIAL_PORT_2 -1
#if ENABLED(TFT_LVGL_UI)
   #define MKS_WIFI_MODULE // MKS WiFi module
#endif
If you do not use the wifi function:
#define SERIAL_PORT 3
#define SERIAL_PORT_2 1
#if ENABLED(TFT_LVGL_UI)
   //#define MKS_WIFI_MODULE // MKS WiFi module
#endif

hakimio, 2021-01-23:
@MKS-Sean it might be a good idea to update documentation (wiki pages).

MKS-Sean, 2021-01-23:
@hakimio Yes, ready to modify it, thank you

closed 2021-01-28 by MKS-Sean