makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #155: How to setup this firmware? - Beginner level

andriusmot, 2021-01-17:
Hi, I'm messing up with my Sapphire Plus and decided to go with marlin firmware for more advanced BLtouch settings. I'm a total beginner in setting up firmware and have lots of questions.
I did setup Visual studio code, I'm able to compile a firmware using marlin example configs and I'm able to boot it on my printer. 
I would like to use this firmware you guys provide, so my questions are:

- Does anyone has dialed in configuration files so printer works as it is intended? (so I can take it from there)
- Where I have to insert these commands? If I do it in beginning of the code, I get error message saying that only one LCD controller has to be selected.

#define TFT_LVGL_UI_FSMC

//#define TFT_LVGL_UI_SPI

If to get an answer you need more info I can provide it tomorrow.

Looking forward to solve this, thanks :)

themacboy, 2021-01-18:
I think that options are dsicontinuated in alst updates, no?

Missatge de andriusmot <notifications@github.com> del dia dg., 17 de gen.
2021 a les 23:50:

mks-viva, 2021-01-18:
@andriusmot 
Hi! You need tell me, what motherboard and display you use?

andriusmot, 2021-01-18:
Hey guys, so today Im going thought the process again. I took your firmware, changed 
`#define MOTHERBOARD BOARD_MKS_ROBIN_NANO_v2` 
to 
`#define MOTHERBOARD BOARD_MKS_ROBIN_NANO ` 

My board: Robin nano V1.2
Screen: #define MKS_ROBIN_TFT35 (not sure if its this one, but its the one that compiles successfully and its commonly used in nano v1.2)

So at this point, it compiles. But it will not boot. It freezes at this point.
[image]

I've also tried using your firmware with Marlins example configs without changing anything, and that works, but it only shows usual marlin interface.

Maybe someone could layout step by step what you do to compile it successfully?

mks-viva, 2021-01-19:
Hi! Provide your firmware to me and i will test it.

andriusmot, 2021-01-19:
> Hi! Provide your firmware to me and i will test it.

 Hey, its master branch im using. Im compiling it with VS Code, platformio and Auto Marlin build plugins are installed.

[Mks-Robin-Nano-Marlin2.0-Firmware-master.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5838077/Mks-Robin-Nano-Marlin2.0-Firmware-master.zip)

mks-viva, 2021-01-20:
You need config it:
1. #define MOTHERBOARD BOARD_MKS_ROBIN_NANO
2. Disable #define MKS_TS35_V2_0 and enable #define MKS_ROBIN_TFT35

andriusmot, 2021-01-20:
Thanks for checking it, but I did those 2. Like I said before I'm getting it compiled and booted, but cinfig file don't seem to be tuned in for Sappire Plus, for example, stepper drivers are set to A4988 instead of TMC2208. So even though printer takes commands, it's not able to execute it. Example configs straight from marlin firmware has ot dialed in, but it gives that old marlin interface, witch sux.

mks-viva, 2021-01-20:
If you use A4988, you need not define driver type and use it directly.

closed 2021-01-28 by MKS-Sean