makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #156: G28 problems with Lvgl UI and 3DTouch

Trvorts, 2021-01-22:
### Description

In the new updates, the G28 command does not work as expected - using 3DTouch, Lvgl ui, wifi module and filament runout sensor.
Used board - MKS Robin Nano V2.

The Z axis does not lift when retesting the homing and the 3Dtouch crashes.

### Steps to Reproduce

1. Select options:
   #define BLTOUCH
  #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
  #define Z_SAFE_HOMING
  #define AUTO_BED_LEVELING_BILINEAR
  #define TFT_LVGL_UI
  #define TOUCH_SCREEN
 #define NUM_SERVOS 1

2. Compile the .bin file and copy the Assets folder, Robin_nano35.bin and MksWifi.bin to upload to the board.

3. In the Touch screen select home all axis option

**Expected behavior:** [Homing run normaly]

**Actual behavior:** [Don't raise the Z axis and the 3D Touch crashes]

#### Additional Information

https://www.youtube.com/watch?v=AUdufma6dhA

[Config.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5856375/Config.zip)

MKS-Sean, 2021-01-23:
Hello, I will test it next Monday

Oranginam, 2021-01-24:
HI,
I have exactely the same problem (with Bltouch V3)
With the previous commit that works perfectly

MKS-Sean, 2021-01-25:
[nano_v2_config.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5866554/nano_v2_config.zip)
@Trvorts @Oranginam 
We used your Trvorts's configuration file to modify some machine parameters. We can return to zero and auto-leveling on our side. I think the key factor is to disable USE_ZMAX_PLUG

Trvorts, 2021-01-26:
> @Trvorts @Oranginam
> We used your Trvorts's configuration file to modify some machine parameters. We can return to zero and auto-leveling on our side. I think the key factor is to disable USE_ZMAX_PLUG

Yes, in the old firmware this option has commented! I'll test and post a feedback. Thankyou so much!

Oranginam, 2021-01-27:
HI
Tested with USE_ZMAX_PLUG commented and .... That work !! :)
Thx

closed 2021-01-28 by MKS-Sean