makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #158: Build fail for Makerbase Nano v 1.2 with TFT35 V 1.0

Bighoneypot, 2021-01-25:
Platformio don't build this repo for my MKS Nano v 1.2 with TFT35 V 1.0

### Description Error

expression must have a constant value -- cannot call non-constexpr function "strstr" (declared at line 48 of "C:\Users\gabri\.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\include\string.h")

#error "TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."
{
	"resource": "/c:/Users/gabri/OneDrive/Documents/REPO/Mks-Robin-Nano-Marlin2.0-Firmware/Marlin/src/inc/SanityCheck.h",
	"owner": "C/C++",
	"code": "35",
	"severity": 8,
	"message": "#error directive: \"TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN.\"",
	"source": "C/C++",
	"startLineNumber": 2467,
	"startColumn": 4,
	"endLineNumber": 2467,
	"endColumn": 9
}

### Steps to Reproduce

Clone this repo : https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware

Build config:
platformio.ini:

default_envs = mks_robin_nano35

Configuation.h:
#define SERIAL_PORT 3
#define MKS_ROBIN_TFT35
#define MOTHERBOARD BOARD_MKS_ROBIN_NANO
#define TFT_LVGL_UI
#define TOUCH_SCREEN

and uncomment this line #define MKS_ROBIN_TFT35 in Configuration.h

and comment this line:

// 480x320, 3.5", SPI Display From MKS
// Normally used in MKS Robin Nano V2
//
//#define MKS_TS35_V2_0

#### Additional Information

MKS-Sean, 2021-01-26:
Hello, you can try the latest update

Bighoneypot, 2021-01-26:
Can you send me link?

MKS-Sean, 2021-01-27:
https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware
@Bighoneypot Pull the latest code of this warehouse

Chlorel, 2021-01-27:
you can try to compile with TMC2209_STANDALONE too ( this way you does not try to use the TMC serial mode )

Bighoneypot, 2021-01-27:
> https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware
> @Bighoneypot Pull the latest code of this warehouse

I have try this, but Platformio don't build, see my img
Loop under build
[image]

MKS-Sean, 2021-01-28:
@Bighoneypot Hello, I can’t reproduce your problem, I think you can provide me with the cnfiguration.h and configuration_adv.h files

Bighoneypot, 2021-01-30:
[configdoth and config_advdoth.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5898369/configdoth.and.config_advdoth.zip)

Now I have this problem:

1. every time the printer is turned on, the display requests calibration
2. after Update asset folder  and .bin old setting are always present
3. if I send the print via wifi, the printer starts immediately and does not wait for the bed and the extruder to be hot
4. and the printer does not print anything because the z axis rises by 3 cm and then prints in the air

I need a info: there is a procedure to clean all eeprom in the Robin Nano v 1.x ???

I need to use my ender 3 pro printer

Chlorel, 2021-01-30:
Don't forget to make a backup of the parameters, once the screen calibration is done (this will prevent it from asking for it again at the next reboot) config->Eeprom Set->Save settings

velkhor, 2021-01-30:
> Platformio don't build this repo for my MKS Nano v 1.2 with TFT35 V 1.0
> ### Description Error
>
> expression must have a constant value -- cannot call non-constexpr function "strstr" (declared at line [...]

Try change 
#define X_DRIVER_TYPE section

My config:
#define X_DRIVER_TYPE  A4988
#define Y_DRIVER_TYPE  A4988
#define Z_DRIVER_TYPE  A4988
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE A4988
#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988

Bighoneypot, 2021-01-31:
I 
[New_configuration.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5899900/New_configuration.zip)

With this configuration Platformio build the .bin, but WiFi is Wrong and printer don't see  .gcode into a Micro Sd.

velkhor, 2021-02-01:
send me file pins_MKS_ROBIN_NANO.h
I need to check this, i think you try doing impossible :), but let me verify my suspections

Bighoneypot, 2021-02-01:
@velkhor thank you, I send you the file

[pins_MKS_ROBIN_NANO.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5905112/pins_MKS_ROBIN_NANO.zip)

velkhor, 2021-02-01:
Do you make hardware upgrade with marlin nano?
i talk about this exactly
https://imgur.com/gallery/duvCECL
this red linea are wires, you need to solder it to drivers

Bighoneypot, 2021-02-01:
@velkhor this is my board
[image]

solution for wifi is welder the drivers?

velkhor, 2021-02-01:
If i see good you have board MKS Robin nano v.1.2, this board don't support TMC_UART, you can use TCM2209 in "classic" mode.
If you want to use UART mode you need to make some hardware changes, but there are no free pins. You need to waste wifi pins. Lost wifi but have TCM in uart mode.

I have nano 1.2 i don't use UART, because i need wifi :)

I try to help you, but need more info :).
Which printer do you have, you upgrade some stock printer or build your own?
Do you have datasheets for drivers (I need to kon what current they have)

Bighoneypot, 2021-02-01:
@velkhor 
Hi, i have an ender 3 pro. I am replacing the original board with this Makerbase nano v1.2 and the tft 35 v 1.0 display. I don't want to lose wifi, for uart it is not necessary, the important thing is that the printer is silent and works as well as with the original card. The stepper motors are creality 42-34 

thank you for support

velkhor, 2021-02-01:
I have ender 3 (not pro).

All four steppers are same? X,Y,Z,E ??
Ender 3 Pro have 42-34 on X,Y,Z
and 42-40 on Extruder
42-34 should use 0.7A
42-40 should use 0.9A

you need to set correct value using multimeter
here is example
https://wiki.fysetc.com/Silent2209/

here you can calculate voltage for steppers
https://learn.watterott.com/silentstepstick/faq/

42-34 = 0.99V
42-40 = 1,27V

As I remember TMC2209 have 1,5A default, it's to big for Ender steppers!!!!

When you calculate current let me know. I give you next steps, what to do ;)

Bighoneypot, 2021-02-01:
@velkhor 
Ho ender 3 (non pro).
All four steppers are same? X,Y,Z,E ??
Ender 3 Pro have 42-34 on X,Y,Z
and 42-40 on Extruder
**Is the same**

My parameters is:

42-34 = 0.725V
42-40 = 0.900V

Tutti e quattro gli stepper sono uguali? X, Y, Z, E ??
Ender 3 Pro ha 42-34 su X, Y, Z
e 42-40 su Extruder
42-34 dovrebbe usare 0.7A
42-40 dovrebbe usare 0.9A

devi impostare il valore corretto usando il multimetro (spero che tu possa usarlo;)
ecco un esempio
https://wiki.fysetc.com/Silent2209/

qui puoi calcolare la tensione per gli stepper
https://learn.watterott.com/silentstepstick/faq/

42-34 = 0,99 V
42-40 = 1,27 V.

Come ricordo che TMC2209 ha 1,5 A di default, è troppo grande per gli stepper Ender !!!!

Quando calcoli la corrente fammelo sapere. Ti do i prossimi passi, cosa fare;)

velkhor, 2021-02-01:
You configure Current with multimerer NOT in files configuration.h?
I'm asking for sure, because you change lines   #define X_CURRENT (in classic mode it don't work)

If yes You now have two ways to do next steps.
1. Use MKS firmware (based on Marilin.
2. Use Marilin.

In 1'st option there is advance configuration LCD menu.
have calssik jerk (not junction deviation)
don't have S-Curve Acceleration
don;t have Linear Advance

In 2'nd option you can enable everything, but need to complie Marilin :)

I choose 2'nd it isn't very hard to do it ;)

Bighoneypot, 2021-02-01:
@velkhor 

I have set up current with multimeter and not in configuration.h

My current value is ok??
Can you send me link for next step??

velkhor, 2021-02-01:
OK. That's ok. :)

GIve me some time, i make config's for you

Bighoneypot, 2021-02-01:
> OK. That's ok. :)
> 
> GIve me some time, i make config's for you

Thank you @velkhor

velkhor, 2021-02-01:
Configuration.h

I set it 700, because 200 is to small 4 ender
#define EXTRUDE_MAXLENGTH 700

Pid autotune (I change it, based on my tune, Ender 3 and 3 pro have same hotend, and this values should be similar)
    #define DEFAULT_Kp  20.84
    #define DEFAULT_Ki   1.96
    #define DEFAULT_Kd  55.47

I dsable this, because I think you don't add extra max z-stop
#define USE_ZMAX_PLUG

drivers section should look that
#define X_DRIVER_TYPE  TMC2209_STANDALONE
#define Y_DRIVER_TYPE  TMC2209_STANDALONE
#define Z_DRIVER_TYPE  TMC2209_STANDALONE
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209_STANDALONE
#define E1_DRIVER_TYPE TMC2209_STANDALONE
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988

steppers, first three values are physically calculated and shouldn't be changed!! Don't listen people about calibrate X,Y,Z steps.
But last value (Extruder) you should be calibrated, and recalibrated when you change extruder, stepper, extruder gear...
Default ender have 95, but you need to calibrate it very very very precise ;)
i have 98,5
https://mattshub.com/blogs/blog/extruder-calibration
#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 400, 95 }

This is your version
#define DEFAULT_MAX_ACCELERATION      { 500, 500, 5, 25 }
last two values are big mistake. Z-axis acceleration have default 100, and E-axis 500
, but I think you can use 1000 without problem. I have 10000, but i change stepper.
#define DEFAULT_MAX_ACCELERATION      { 500, 500, 100, 1000 }

You have defined classik Jerk, but i see default values. My ender have 8,8,0.4
#define CLASSIC_JERK

In my config i use Junction_deviation, but correct value for ender is 0.08 (0.013 - default MKS value)
#define JUNCTION_DEVIATION_MM 0.08 // (mm) Distance from real junction edge

After upload new firmware, set manually head in half off all axes and check motor direction.
Propably these values are correct... but if not change false/true

// Invert the stepper direction. Change (or reverse the motor connector) if an axis goes the wrong way.
#define INVERT_X_DIR true
#define INVERT_Y_DIR true
#define INVERT_Z_DIR false

// @section extruder

// For direct drive extruder v9 set to true, for geared extruder set to false.
#define INVERT_E0_DIR true
#define INVERT_E1_DIR false
#define INVERT_E2_DIR false
#define INVERT_E3_DIR false
#define INVERT_E4_DIR false
#define INVERT_E5_DIR false
#define INVERT_E6_DIR false
#define INVERT_E7_DIR false

If you need in future Runout sensor, this command should be correct
  #define FILAMENT_RUNOUT_SCRIPT "M25"
not
  #define FILAMENT_RUNOUT_SCRIPT "M600"

I speed up z-homing (4*60*4). Standard value (4*60) are very slow.
#define HOMING_FEEDRATE_MM_M { (50*60), (50*60), (4*60*4) }

Check this, but in my opinion this should be comented (this rotate screen upside down
#define TFT_ROTATION TFT_ROTATE_180

Configuration_adv.h
I change this for fastest homing, at end.
#define HOMING_BUMP_MM      { 4, 4, 2 }       // (mm) Backoff from endstops after first bump
#define HOMING_BUMP_DIVISOR { 5, 5, 4 }       // Re-Bump Speed Divisor (Divides the Homing Feedrate)

If you uncomment this, you enable line advance. Read about this, this is good feature, but now I sugest leave this commented.
//#define LIN_ADVANCE

I leave default values showed bellow, because this is for UART ;)
#define X_CURRENT
#define Y_CURRENT
#define Z_CURRENT
#define E0_CURRENT
//#define MONITOR_DRIVER_STATUS
//#define HYBRID_THRESHOLD

I left default pins_MKS_ROBIN_NANO.zip, because you make bad changes ;)

Here is file 4 U.
https://we.tl/t-A5oyo5jEWP
I can't attach it because it is to big

BTW if you have bad adhesion, try increase hotend temp at first layer. I change it 10 deg up, and adhesion is good. Earlier break from table after 10-12 layers.

Bighoneypot, 2021-02-01:
@velkhor 
so now I do a build of the folder you sent me without changing other parameters?

Bighoneypot, 2021-02-01:
@velkhor 
I have to try your project, but If send a print over wifi, but starts without heating the bed and the extruder

velkhor, 2021-02-01:
This is always set in g-code, because printer doesn't know what filament you load ;)

You should set it with begin of gcode

add in Cura to machine settings Start G-code
M190 S[first_layer_bed_temperature] ; wait for bed temp
M109 S[first_layer_temperature] ; wait for extruder temp

If this don't work senc me gcode

Bighoneypot, 2021-02-01:
@velkhor don't work 
[image]

velkhor, 2021-02-01:
Sorry My mistake, Cura don't use temp in start g-code. They are generated automatically from cura settings
[image]

In gcode should be somenthink like:

;Generated with Cura_SteamEngine 4.7.1
M140 S50 // this set bed temp to 50 deg (but interpreting of gcode run)
M105 // this report temp to host https://marlinfw.org/docs/gcode/M105.html, both options are enabled
M190 S50 // this wait for bed temp 50 deg
M104 S200 // this set bed 200 deg
M105 // this report temp to host https://marlinfw.org/docs/gcode/M105.html, both options are enabled
M109 S200 // this wait until extuder have 200 deg
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
.... rest of code

Try this gcode, it's generated from pursa slicer, but should work, it uses few cm of filament ;)

[leveling.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5907046/leveling.zip)

this g-coide make some moves before heat up bed, dont be afraid, and wait until it print.

Bighoneypot, 2021-02-02:
your G-Code file was printed well, but display don't show gcode preview.
[image]

@velkhor 
But with my stl e my cura setup don't work fine.
If send file , my print start immediately

[image]

[CE3PRO_ticwatch_v2.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5909824/CE3PRO_ticwatch_v2.zip)

velkhor, 2021-02-02:
Yeep. I agree don't show preview, because there is no preview :)

In my opinion your g-code looks good, but i check it on my printer at evening.

Bighoneypot, 2021-02-02:
Ok, perfect. Thanks

velkhor, 2021-02-02:
Hi
I don't know what I can say.
There is everything ok. My printer start heat bed to 50 deg, next heat up nozzle to 227 deg, and start print.
Here are photos ;)
[image]
[image]
Do You print it using PLA? My PLA boil at start ptint, i lower temp to 207 deg.
Printed id 24 mins. I sped up it 200%, but I have a lot of modification.

velkhor, 2021-02-02:
Do you compile using VisualStudio, or Arduino?

velkhor, 2021-02-02:
One more option, I build it 4U test it.

[Robin_nano35.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5914349/Robin_nano35.zip)

Bighoneypot, 2021-02-03:
> Do you compile using VisualStudio, or Arduino?

Visual Studio

Bighoneypot, 2021-02-03:
> One more option, I build it 4U test it.
> 
> [Robin_nano35.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5914349/Robin_nano35.zip)

@velkhor  Now I try this

velkhor, 2021-02-03:
Can you record some movie?
run my g-code leveling.zip
i need see on screen LCD, and print head.

Bighoneypot, 2021-02-03:
@velkhor your g-code leveling.zip run correctly. Problem is when i send with wifi my gcode.

With your g-code levelig print is ok.

link for video :
https://drive.google.com/drive/folders/1eitRDgrrWIhOabOTfbcfJFU3gI28UPRQ?usp=sharing

velkhor, 2021-02-03:
No. It isn't. Nozzle show 18 deg, and bed 20 deg. There should no printing.

velkhor, 2021-02-03:
Can you verify your connections to bead and heater? Are they correctly connected?

Bighoneypot, 2021-02-03:
Yes are correct. With your file print is ok

Bighoneypot, 2021-02-03:
> Can you verify your connections to bead and heater? Are they correctly connected?

@velkhor  are ok

velkhor, 2021-02-03:
Check messenger. ;)

Bighoneypot, 2021-02-03:
@velkhor messenger??

Bighoneypot, 2021-02-16:
@velkhor for this moment all work fine. Thank you @velkhor

Asta7180, 2023-12-28:
hi,
In configuration.h
when I put: #define USE_YMIN_PLUG

I get an error telling me that I have to use: #define USE_YMAX_PLUG

but for my printer it is good: USE_YMIN_PLUG

velkhor, 2023-12-28:
hi,
in default this section looks that

#define USE_XMIN_PLUG
// #define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_IMIN_PLUG
//#define USE_JMIN_PLUG
//#define USE_KMIN_PLUG
//#define USE_XMAX_PLUG
#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
//#define USE_IMAX_PLUG
//#define USE_JMAX_PLUG
//#define USE_KMAX_PLUG

but if you want use Y MIN and don't use Y MAX

you need to comment Y MAX too

and should look that

#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_IMIN_PLUG
//#define USE_JMIN_PLUG
//#define USE_KMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
//#define USE_IMAX_PLUG
//#define USE_JMAX_PLUG
//#define USE_KMAX_PLUG

Asta7180, 2023-12-28:
> hi, in default this section looks that
>
> #define USE_XMIN_PLUG // #define USE_YMIN_PLUG #define USE_ZMIN_PLUG //#define USE_IMIN_PLUG //#define USE_JMIN_PLUG //#define USE_KMIN_PLUG //#define [...]

Thank you for your answer, but I already tested, and I got the error message:

#error 'Enable USE_YMAX_PLUG when homing Y to MAX.'

my config:
// @section homing

// Specify here all the endstop connectors that are connected to any endstop or probe.
// Almost all printers will be using one per axis. Probes will use one or more of the
// extra connectors. Leave undefined any used for non-endstop and non-probe purposes.
#define USE_XMIN_PLUG
#define USE_YMIN_PLUG
#define USE_ZMIN_PLUG
//#define USE_IMIN_PLUG
//#define USE_JMIN_PLUG
//#define USE_KMIN_PLUG
//#define USE_XMAX_PLUG
//#define USE_YMAX_PLUG
//#define USE_ZMAX_PLUG
//#define USE_IMAX_PLUG
//#define USE_JMAX_PLUG
//#define USE_KMAX_PLUG

velkhor, 2023-12-28:
OK. Now I see :)

This error is generated by wrong homing :)

I suppose you need to change homing direction too.
in default:
#define Y_HOME_DIR 1

after change endstop should look

#define Y_HOME_DIR -1

Asta7180, 2023-12-28:
> OK. Now I see :)
> 
> This error is generated by wrong homing :)
> 
> I suppose you need to change homing direction too. in default: #define Y_HOME_DIR 1
> 
> after change endstop should look
> 
> #define Y_HOME_DIR -1

Awesome!
Thanks I didn't look at this line