makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #160: Transfer of files by Wifi with Cura is wrong.

pistachin2, 2021-01-26:
Since I have Marlin, the transfer of files from Cura by Wifi does not work well, the files I send are incompetent or corrupt.
It happens to me with the Nano V1.2 and with the V2.
If the Nano V1.2 is loaded with its own firmware 2.0.3.5 or 2.0.3.4, the transfer works perfectly.
Do you happen to someone else?

pistachin2, 2021-01-28:
Hello, I have been able to verify that with the MKS Marlin (downloaded today 01-28) what happens to the module is that it disconnects from the printer every 15 seconds. In Cura you can check, the connection message appears every little bit.
It happens to me in a Nano V1.2 and in a V2, but with V1.2 with its own fimware it does not happen, it is something that has me very disturbed.
Thanks in advance for whoever is interested.

Bighoneypot, 2021-01-31:
> Since I have Marlin, the transfer of files from Cura by Wifi does not work well, the files I send are incompetent or corrupt.
> It happens to me with the Nano V1.2 and with the V2.
> If the Nano V1.2 is [...]

Same problem with robin nano v1.x, tft 35 v1.0 and wifi

[image]

MKS-Sean, 2021-02-03:
@Bighoneypot The wifi module has not been configured yet, can you find the wifi hotspot after waiting for a while?

MKS-Sean, 2021-02-03:
@pistachin2 Which version of your cura plugin is, can you send it to me? And what is your cura version? I will continue to test it

Bighoneypot, 2021-02-03:
@pistachin2 Ultimaker Cura version 4.8.0
MKS WIFI PLUGIN 1.1.2

pistachin2, 2021-02-03:
@MKS-Sean Hello, using the Cura marketplace plugin, until today it was 1.1.2, and today it has been updated to 1.2.0, but it still does not work.
The most curious thing is that the Nano v1.2 with its own firmware does work, but if I put the MKS Marlin it does not work. In the Nano V2 (which is Marlin yes or yes) it does not work either.
[image]

pistachin2, 2021-02-03:
@MKS-Sean From what I've seen, in Cura the printer connected message appears every little time (about 40-50 seconds), without doing anything.
And when some printing is sent, the ping to the MKS WIFI IP is lost.

Chlorel, 2021-02-15:
I have the same kind of small inconveniences, when I want to transfer a small file, it works from time to time, but as soon as the file is bigger, I can no longer transmit it via wifi, so I have to transfer it directly to the sd card.

I use cura 4.6.1 with plugin 1.1.2

I tried to update MKSWifi (the one provided in the github of MKS Robin E3P) without more success, and as I can't set up a configuration under visual studio to compile it, I'm a bit stuck. Maybe @MKS-Sean will find us a miracle solution after the chinese new year ?

Info: with the original firmware provided with my printer ( Wanhao D12 ) it works ! But with other hint's ( like not showing preview after file transfert only showing when restarting the print from printer or from SD Console in Cura )

pistachin2, 2021-02-26:
Hello @MKS-Sean , have you been able to investigate the problem? Do you need us to report any information to you or to carry out a test?
The summary that I can do of all the tests (under the MKS Marlin and with Nano V1.2 and Nano V2) is:
- With very small files there are times that it works, but very rarely.
- The files that are transmitted, when trying to open them with Notepad ++ are corrupted and the file will not be opened.
- Every 40 or 50 seconds in Cura the printer connected message appears.
- The commands that are sent from Cura (movements, homing) take about 10 or 15 seconds to execute, but they are not always executed.

As I have commented in other messages, in Nano V1.2 and its own firmware, it works without any problem.

Bighoneypot, 2021-02-27:
Hi Guys, the problem is present. this happens: if I print via Wifi, the print starts immediately and does not wait for the nozzle and the bed to be warm. So I have to turn off the print, put the file on the card and then print directly from the card. If I print directly with the wifi, everything doesn't work