makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #164: Отображает не полностью кнопки.

Vycheslav-rus, 2021-02-10:
Привет. Я нарисовал свои картинки, чтобы изменить внешний вид. Делал всё как в описании: загрузил их в конвертер и преобразовал в bin файлы. Но после загрузки в принтер они отображаются не полноценно. А когда их загружаешь в MKS TOOLS они отаброжаються корректно. 
Плата MKS ROBIN NANO V2
[image]
[image]
[assets.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5958433/assets.zip)

artdezigner, 2021-02-11:
Привет, по всей видимости MKS TOOLS работает плохо, я убил где-то час времени отрисовав все картинки и закинув их в MKS TOOLS, подбирая вручную картинки по смыслу, а в итоге они также криво подгрузились. Если нужно - вот исходники:
[Исходники картинок.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/5968649/default.zip)

artdezigner, 2021-02-16:
Что, поддержка MKS отвалилась? АУ!

MKS-Sean, 2021-02-21:
@Vycheslav-rus @artdezigner Здравствуйте, наш текущий инструмент не поддерживал LVGL UI, вы можете использовать веб-инструмент для преобразования иконок
https://lvgl.io/tools/imageconverter

Vycheslav-rus, 2021-02-22:
Привет. Я всё сделал как в описание:
https://lvgl.io/tools/imageconverter
Choose color format:True color.
Choose file output format:Binary RGB565.
Но картинки не отаброжаються.
[image]
Вот исходник, проверьте сами пожалуйста.
[assets+jpg.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6022211/assets%2Bjpg.zip)

Vycheslav-rus, 2021-02-22:
Привет. Мои картинки отконвертированны правильно.
Неправильное их отображение связано с прошивкой. 
Я провел эксперимент, взял одну из картинок и переименовал её.
Чтобы картинка загрузилась я прописала в pic_manager.cpp

[image]

Потом в файле draw_tool.cpp добавил строки для отображения моей картинки в свободном месте экрана.

[image]

При прошивке принтера, все картинки загружались в алфавитном порядке, кроме моей.
Моя корзинка загрузилась последней, после загрузки шрифтов.
И вот что вышло, она отобразилась правильно.

[image]

Проблема установки не родных картинок связана с прошивкой, не правильное распределение памяти. Прошивка загружает картинки на их старое место, где бала родная картинка. И если новая картинка больше по объему, то её обрезает, а если она меньше то к ней добавляется хаос от старой картинки.
Мой эксперимент подтвердил этот факт, прошивка загрузила мою картинку в другое место память, куда-то после шрифтов. И она осталась неповреждённой.