makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #165: Custom menu problem

artdezigner, 2021-02-11:
Добрый день, плата Robin Nano V2 + TS35. Обновил прошивку и пропало дополнительное меню. Прошу помочь. В файлах
Good afternoon, Robin Nano V2 + TS35 board. I updated the firmware and the additional menu disappeared. Please help me. In the files
configuration.h:
//// Preheat Constants
#define PREHEAT_1_LABEL       "ABS"
#define PREHEAT_1_TEMP_HOTEND 240
#define PREHEAT_1_TEMP_BED    95
#define PREHEAT_1_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_2_LABEL       "PET"
#define PREHEAT_2_TEMP_HOTEND 230
#define PREHEAT_2_TEMP_BED    70
#define PREHEAT_2_FAN_SPEED     0 // Value from 0 to 255

#define PREHEAT_3_LABEL       "PLA"
#define PREHEAT_3_TEMP_HOTEND 180
#define PREHEAT_3_TEMP_BED     60
#define PREHEAT_3_FAN_SPEED     0 // Value from 0 to 255
////

configuration_adv.h:
/**
 * User-defined menu items that execute custom GCode
 */
#define CUSTOM_USER_MENUS ////
#if ENABLED(CUSTOM_USER_MENUS)
  //#define CUSTOM_USER_MENU_TITLE "Custom Commands"
  #define USER_SCRIPT_DONE "M117 User Script Done" ////
  #define USER_SCRIPT_AUDIBLE_FEEDBACK ////
  //#define USER_SCRIPT_RETURN  // Return to status screen after a script
									   
  #define USER_DESC_1 "Preheat for " PREHEAT_1_LABEL
  #define USER_GCODE_1 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  #define USER_DESC_2 "Preheat for " PREHEAT_2_LABEL
  #define USER_GCODE_2 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_2_TEMP_HOTEND)

  #define USER_DESC_3 "Preheat for " PREHEAT_3_LABEL
  #define USER_GCODE_3 "M140 S" STRINGIFY(PREHEAT_3_TEMP_BED) "\nM104 S" STRINGIFY(PREHEAT_3_TEMP_HOTEND)

  #define USER_DESC_4 "Cool"
  #define USER_GCODE_4 "M108\nM106 255"

  #define USER_DESC_5 "Hotend"
  #define USER_GCODE_5 "M104 S" STRINGIFY(PREHEAT_1_TEMP_HOTEND)

  #define USER_DESC_6 "Bed"
  #define USER_GCODE_6 "M140 S" STRINGIFY(PREHEAT_1_TEMP_BED)
#endif

Chlorel, 2021-02-15:
I think you have to reattach the "draw_more.cpp" module in the "draw_tool.cpp" menu because from what I see it has been detached (I can't tell you more about it)

artdezigner, 2021-02-16:
Hmm, it's strange that in the previous version of Marlin from MKS, everything was also configured and the menu was on the screen. Now it just doesn't exist, why do that? Where is MKS support? By the way, even with the cooling of the hotend, the problem was not solved, although I previously wrote here how to do it. In principle, almost nothing has changed in the quality of support since Robin Nano v. 1. The quality of support and firmware is poor.
/
Хм, странно, что в предыдущей версии Марлин от MKS все было также настроено и меню было на экране. Теперь же его просто нет, зачем так делать? Где поддержка MKS? Кстати даже с охлаждением хотэнда проблему не решили, хотя я ранее здесь написал как это исправить. В принципе почти ничего не изменилось со времени Robin Nano v.1. Качество поддержки и прошивок все также плохое.
/

MKS-Sean, 2021-02-21:
@Chlorel @artdezigner Hello, I have repaired this problem in the latest submission.

Chlorel, 2021-02-21:
@MKS-Sean I had modified it for the firmware I use on my printer as well. But as that part was deleted, or suspended, I thought there must be a reason and didn't propose the modification.

MKS-Sean, 2021-02-21:
@Chlorel Okay, thank you

artdezigner, 2021-02-24:
Thank you so much, the menu is working! MKS, please when releasing new firmware, do not break what already worked. By the way, the fan does not work in the latest firmware, but I wrote about this solution [here](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/issues/142#issue-758025370)

Огромное спасибо, меню заработало! MKS, пожалуйста при выпуске новых прошивок не ломайте то, что уже работало. Кстати в последней прошивке так и не работает вентилятор хотэнда, но я писал об этом решении.