makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #167: [Robin Nano 2.0] file does not print when it is in a sub-directory

ikkez, 2021-02-13:
I just tried latest firmware today and found a new weird issue. When the .stl file is not on the root of the sdcard, but in a sub-folder, it does not start printing.

### Steps to Reproduce

1. put stl file on root of sdcard and also copy it into a folder
2. try printing both files

**Expected behavior:** both files are printing

**Actual behavior:** only file at root directoy is starting to print

MKS-Sean, 2021-02-21:
Hello, are you talking about gcode files? Can it be heated normally when printing files in subfolders? I tested it here, the sub-file printing is normal printing, you can send your configuration file and gcode file to us for testing

ikkez, 2021-03-02:
yes I mean gcode files. I can select the file but it does not start to heat up and just stays in the print view forever. 
My SD Card is formatted as FAT32 (8GB total). 

My gcode starts with: 

```gcode
;FLAVOR:Marlin
;TIME:2085
;Filament used: 0.829538m
;Layer height: 0.12
;MINX:-28.939
;MINY:-33.942
;MINZ:0.3
;MAXX:38.939
;MAXY:33.942
;MAXZ:11.58
;Generated with Cura_SteamEngine 4.8.0
M140 S40
M105
M190 S40
M104 S208
M105
M109 S208
M82 ;absolute extrusion mode
G28 ;Home
G1 Z15.0 F6000 ;Move the platform down 15mm
;Prime the extruder
G92 E0
G1 F200 E3
G92 E0
G92 E0
G92 E0
G1 F1500 E-6.5
;LAYER_COUNT:95
;LAYER:0
M107
G0 F4800 X-28.903 Y1.213 Z0.3
;TYPE:SKIRT
G1 F1500 E0
G1 F1200 X-28.939 Y-.255 E0.07546
G1 X-28.937 Y-.771 E0.10197
...
```

my current config.
https://github.com/ikkez/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/Configuration.h
https://github.com/ikkez/Mks-Robin-Nano-Marlin2.0-Firmware/blob/master/Marlin/Configuration_adv.h