makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #168: I set TFT_ROTATION to TFT_ROTATE_180 in Robin Nano v1.2, but it appears as TFT_MIRROR_Y.

Vagabond-K, 2021-02-14:
### Description

I set TFT_ROTATION to TFT_ROTATE_180 in Robin Nano v1.2, but it appears as TFT_MIRROR_Y.
TFT is controlled by FSMC.

### Steps to Reproduce

1. #define TFT_ROTATION TFT_ROTATE_180
2. Build and Update.

**Expected behavior:** TFT rotated 180 degrees.

**Actual behavior:** TFT is reversed to Y axis.

#### Additional Information

* None

MKS-Sean, 2021-02-21:
[image]
Hello, the TFT in my test can rotate 180 degrees normally, but the touch value is wrong, I have been correcting it in the latest commit, the following is the configuration file I use.
[Nano-V1.2-TFT-rotation-180.zip](https://github.com/makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware/files/6017383/Nano-V1.2-TFT-rotation-180.zip)

Vagabond-K, 2021-02-27:
> [image]
> Hello, the TFT in my test can rotate 180 degrees normally, but the touch value is wrong, I have been correcting it in the latest commit, the following is the configuration file I use. [...]

Thank you very much. My screen rotated well.

Vagabond-K, 2021-02-27:
Thank you very much. My screen rotated well.

closed 2021-02-27 by Vagabond-K