makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #172: BEEP when I plug the Thermistor & the E0 & E1 won't work

adrianbelano, 2021-02-19:
Hi, anyone know why? 

1st problem: When I plug the thermistor, when it starts, BEEP sound.

2nd problem: When I change the NUM_Z_STEPPER_DRIVERS 2, nothing changed.

Is there any problem with the board or the software?

MKS-Sean, 2021-02-21:
Hello, what motherboard do you use？
1. It is necessary to check whether the thermal sensor is damaged, and the temperature alarm is triggered when the main board is connected
2. E0 extrusion needs to reach a certain temperature in order to be extruded normally, you need to check whether the temperature is reached

adrianbelano, 2021-02-25:
Thank you for the answer. The thermistor is damaged, I changed it and fwala, it fixed.

closed 2021-02-25 by adrianbelano