makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #173: WIFI Support for maril UI Mode ...

themacboy, 2021-02-20:
Good morning,

I never liked the MKS UI, and sincerelly prefer the Marlin color UI.

Can you add support for wifi on Marlin?

Thanks in advance !

[image]

MKS-Sean, 2021-02-21:
Yes, we are happy to do this, I need to evaluate the difficulty first

pistachin2, 2021-02-26:
Hi, if I'm not wrong, the marlin options WIFISUPPORT and ESP3D_WIFISUPPORT only work on boards that have the ESP32 microprocessor, and the Robin Nano use STM32.

Darkraven2k3, 2021-03-21:
Any Updates cause e3d Fork for mkswifi esp8266 is working

https://github.com/luc-github/ESP3D/tree/3.0

themacboy, 2021-05-05:
> Yes, we are happy to do this, I need to evaluate the difficulty first

@MKS-Sean : Any new about that?

themacboy, 2021-05-05:
> Any Updates cause e3d Fork for mkswifi esp8266 is working
> 
> https://github.com/luc-github/ESP3D/tree/3.0

Please do you know any tutorial to install ESP3D in a mks board? Thx in advance

just-trey, 2022-01-21:
+1 if this is possible. I have tried with no luck.