makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #176: 3D Touch with Robin nano E3P

Chlorel, 2021-02-21:
I tried to install a Robin Nano E3P card on an Ender 3 v1 with a TS35 screen, everything goes relatively well until the addition of a 3D Touch (ordered on aliexpress in the Makerbase shop), the printer starts by adjusting the Z Offset ... then starts to palpate the 25 points (I chose BILINEAR with 5 points per axis) and for no apparent reason stops randomly either after 1 test point, or after 5 test points ... But never reached the end of the leveling of the plate ... 
I chose this version of the Marlin source because it seems more advanced than the github version specific to the E3P board. Would I have made a mistake by doing so ?
I would appreciate some research to solve this little problem.

Chlorel, 2021-02-24:
Partially solved by compiling a firmware without auto leveling, adjusting the bed manually a first time, then applying the firmware with auto leveling again !
I don't see what could have happened.
Would it be possible once a BL/3D Touch is installed to keep the 2 modes of tray adjustment? ( Manual and Automatic ) ?

MKS-Sean, 2021-02-24:
3Dtouch can be used instead of BL_TOUCH，We have ender 3 machines here, you can send the firmware bin you are currently using to us for testing.Regarding 3Dtouch information, please confirm whether it can help you:
https://github.com/makerbase-mks/MKS-Robin-Nano-V2.X/wiki/3D_TOUCH