makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #177: После загрузки файла по WIFI принтер завершает работу.

student20102, 2021-02-21:
Обновил прошивку. Установлен модуль PWC. После передачи файла по WIFI принтер выключается как после завершения печати. после включения можно начать печать все работает. как исправить выключение после передачи по WIFI ?

student20102, 2021-02-27:
Updated the firmware. The PWC module is installed. After transferring the file via WIFI, the printer turns off as after finishing printing. after switching on, you can start printing everything works. How to fix shutdown after WIFI transmission? Robon Nano 1.2 board

Ilya-Pak, 2021-03-01:
https://github.com/Sergey1560/fb4s_howto/blob/master/mks_ui/readme.md
Watchdog causes issue. Switch it off until new release fix the bug. More explanation under link.

Сергей указал, что это из-за Watchdog 
//#define USE_WATCHDOG
Отключите и проблема уйдет.

student20102, 2021-03-02:
Если закоментировать сторожевой таймер то файл даже не начинает передаватся просто принтер зависает и все.

Ilya-Pak, 2021-03-02:
Осень странно, у меня сработало, а пины вафли правильно настроены? Три пина IO1 IO2 reset?

Dom749s, 2021-03-02:
For me when I comment
"// # define USE_WATCHDOG"
the transfer is working but the printer freezes at the end of the transfer.

student20102, 2021-03-03:
#define MKSPWC
  #ifdef MKSPWC
    #define SUICIDE_PIN                     PB2   // Enable MKSPWC SUICIDE PIN
    #define SUICIDE_PIN_INVERTING          false  // Enable MKSPWC PIN STATE
    #define KILL_PIN                        PA2   // Enable MKSPWC DET PIN
    #define KILL_PIN_STATE                  true  // Enable MKSPWC PIN STATE
  #endif

  #define MT_DET_1_PIN                      PA4   // LVGL UI FILAMENT RUNOUT1 PIN
 //#define MT_DET_2_PIN                      PE6   // LVGL UI FILAMENT RUNOUT2 PIN
  #define MT_DET_PIN_INVERTING             false  // LVGL UI filament RUNOUT PIN STATE

  #define WIFI_IO0_PIN                      PC13  // MKS ESP WIFI IO0 PIN
  #define WIFI_IO1_PIN                      PC7   // MKS ESP WIFI IO1 PIN
  #define WIFI_RESET_PIN                    PA5   // MKS ESP WIFI RESET PIN

Ilya-Pak, 2021-03-03:
Странно у меня на робине нано 1.1 другие пины... у меня медведь Летающий 4s.
0 - PA8
1 - PC7
RST - PA5

student20102, 2021-03-03:
Для нано 1.2 так правильно. Обидно что до обновы все работало корректно. я по тупости затер ту папку этой версией.

RAFALFRY, 2021-03-11:
Has someone managed to solve this problem?

student20102, 2021-03-13:
> Has someone managed to solve this problem?
NO