makerbase-mks/Mks-Robin-Nano-Marlin2.0-Firmware #179: cannot compile for nano1.2

memelimestan, 2021-02-25:
hi, i cannot compile the marlin for mks robin nano 1.2 and tft35. and this is the error;

Building in release mode
Linking .pio\build\mks_robin_nano35\firmware.elf
c:/users/levo7/.platformio/packages/toolchain-gccarmnoneeabi@1.70201.0/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/bin/ld.exe: cannot open linker script file C:/Users/levo7/desktop/Mks-Robin-Nano-Marlin2.0-Firmware-master/buildroot/share/PlatformIO/ldscripts/mks_robin_nano.ld: Invalid argument
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\mks_robin_nano35\firmware.elf] Error 1

Chlorel, 2021-02-27:
Hello,
I can only suggest you to take a fresh installation of the master branch of this github, to make only the modifications indicated in the readme from this folder: 

    Build config:

    platformio.ini:

    default_envs = mks_robin_nano35

    Configuration.h:
    #define SERIAL_PORT 3
    #define MKS_ROBIN_TFT35
    #define MOTHERBOARD BOARD_MKS_ROBIN_NANO
    #define TFT_LVGL_UI
    #define TOUCH_SCREEN

To make a compilation test with only these modifications ... 
If it doesn't compile it means you have a problem installing your environment, because I can assure you that it compiles, with just 1 warning in the lvgl library.

memelimestan, 2021-02-27:
Something was wrong in my administrator permissions, changed them and violla

closed 2021-02-27 by memelimestan

Gornerrrrr, 2021-10-18:
same problem

Gornerrrrr, 2021-10-18:
Compile successed when no changes.
Compile failed when after the configuration MKS Robin Nano V1.2.